mongoose-os-apps/shelly-homekit #610: Flashing RGBW2 -> no HomeKit available

Splix123, 2021-04-06:
Wanted to flash my **RGBW2** via script with the HomeKit software but it only tells me that i can‘t update from this firmware-version (tried it with _1.9_ and _1.10_). Did i oversee something or is a specific firmware needed?

Thanks in andvance

andyblac, 2021-04-06:
> Wanted to flash my **RGBW2** via script with the HomeKit software but it only tells me that i can‘t update from this firmware-version (tried it with _1.9_ and _1.10_). Did i oversee something or is a specific firmware needed?
> 
> Thanks in andvance

you need zip file then do something like
`./flash-shelly.py --local-file flashfile.zip hostnameofdevice`

Splix123, 2021-04-06:
sorry but what do you mean with "_flashfile.zip_" or "_rgbw-fw-fade.zip_"? Where can I get these? Can't find them in the shelly-homekit folder.

timoschilling, 2021-04-06:
sorry it's a bit confusing at the moment, because RGBW2 support isn't released yet.
1. download https://github.com/mongoose-os-apps/shelly-homekit/suites/2382857483/artifacts/50604020
2. unzip it
3. run `./flash-shelly.py --local-file shelly-homekit-ShellyRGBW2.zip <host name of device>`

Splix123, 2021-04-07:
@timoschilling no need to feel sorry, in fact I feel sorry for bothering you with my stupid problems and I'm really grateful for your help.

Anyways - As soon as I include the "--local-file..." part it starts a Webserver on port 8381 but then proceeds to not find any device. This is strange however because without file attachment it does find the shelly but has no firmware for it.

Btw - really love your work here and appreciate the help!

andyblac, 2021-04-07:
> @timoschilling no need to feel sorry, in fact I feel sorry for bothering you with my stupid problems and I'm really grateful for your help.
>
> Anyways - As soon as I include the "--local-file..." part [...]

can you redo the command but add `--log-file debug.log` to it
ie run
`./flash-shelly.py --local-file shelly-homekit-ShellyRGBW2.zip --log-file debug.log <host name of device>`
and post the debug log file here.

Splix123, 2021-04-07:
```
2021-04-07 16:26:25,213 INFO 938 Starting local webserver on port 8381
2021-04-07 16:26:25,722 INFO 1026 
2021-04-07 16:26:25,723 INFO 1028 Devices found: 0 Upgradeable: 0 Flashed: 0 Failed: 0
2021-04-07 16:26:25,723 INFO 1032 Log file created: debug.log
```

really doesn't say much...

andyblac, 2021-04-07:
> ```
> 2021-04-07 16:26:25,213 INFO 938 Starting local webserver on port 8381
> 2021-04-07 16:26:25,722 INFO 1026 
> 2021-04-07 16:26:25,723 INFO 1028 Devices found: 0 Upgradeable: 0 Flashed: 0 Failed: 0
> 2021-04-07 16:26:25,723 INFO 1032 Log file created: debug.log
> ```
> 
> 
> really doesn't say much...

i need full file please.

Splix123, 2021-04-07:
That IS in fact the whole file. I know weird...

[debug.log](https://github.com/mongoose-os-apps/shelly-homekit/files/6273742/debug.log)

[debug_without_flashfile.log](https://github.com/mongoose-os-apps/shelly-homekit/files/6273749/debug_without_flashfile.log)

andyblac, 2021-04-07:
> That IS in fact the whole file. I know weird...
> 
> [debug.log](https://github.com/mongoose-os-apps/shelly-homekit/files/6273742/debug.log)
> 
> [debug_without_flashfile.log](https://github.com/mongoose-os-apps/shelly-homekit/files/6273749/debug_without_flashfile.log)

hmm ok try
`./flash-shelly.py --local-file shelly-homekit-ShellyRGBW2.zip --log-file debug.log -v5 <host name of device>`

Splix123, 2021-04-07:
Well that gives a proper log

[debug.log](https://github.com/mongoose-os-apps/shelly-homekit/files/6274903/debug.log)

andyblac, 2021-04-07:
> Well that gives a proper log
> 
> [debug.log](https://github.com/mongoose-os-apps/shelly-homekit/files/6274903/debug.log)

thx i'll review it get back to you.

andyblac, 2021-04-07:
> > Well that gives a proper log
> > [debug.log](https://github.com/mongoose-os-apps/shelly-homekit/files/6274903/debug.log)
> 
> thx i'll review it get back to you.

ok found issue`2021-04-08 00:39:34,051 DEBUG 387 File does not exist`

drop you zip in into tools folder and try script.

andyblac, 2021-04-07:
we have just added rgbw2 to alpha1 status so you now try

`./flash-shelly.py --variant alpha1 <host name of device>`

Splix123, 2021-04-07:
Thanks!!! that worked!
But I have to say I did it with the first command because the second one told me to switch modes after the reboot, and my shelly didn't do that it just kept the color mode.

Still much love and thanks for the quick help

closed 2021-04-08 by Splix123