mongoose-os-apps/shelly-homekit #614: Garage door automatic learning

RobbyesHomekit, 2021-04-07:
Good morning, it would be interesting to implement automatic and manual learning like on roller shutters even at the garage door without sensor use. Would that be possible? 
I apologise for my English

BlancoDeMallloca, 2021-04-07:
For the calibration of rollershutter with Shelly 2.5 the firmware measures the power consumption in watt. 

If you use a Shelly 1 in garage door mode there won’t be any power in watt since it is only a simple pulse to trigger the garage door (and Shelly 1 isn't able to measure power on watt). If you are not using a sensor the firmware will tell you that garagedoor is opening/closing depending on the time set in the firmware but it won’t get any input for the status of the garagedoor (opened or closed). That’s why there is a need the close sensor. That’s the logic behind Shelly 1 garage door mode. 

Shelly 2.5 may be different but I do not have this mode in use.

RobbyesHomekit, 2021-04-07:
I apologise if I didn't specify, but I meant shelly 2.5. I am currently using a firmware shelly that allows both the automatic learning and the manual learning to be used by setting closing and opening times into a garage. Since I'm flashing them all with this more stable firmware, I wanted to know if it could be implemented on this one too

timoschilling, 2021-04-07:
Does I understand you right that you want to use a Shelly 2.5 as garage door opener, but you don't want to use a close contact?
And to achieve that you ask for an automatic learning mode?

RobbyesHomekit, 2021-04-07:
The shelly 1 is using it to control the remote control of an automatic gate but I see that without a sensor it does not fulfil opening times and no closing times are present. It works, but animation times are not real. Too bad!

RobbyesHomekit, 2021-04-07:
> Does I understand you right that you want to use a Shelly 2.5 as garage door opener, but you don't want to use a close contact?
> And to achieve that you ask for an automatic learning mode?

Right now my shelly 2.5 homekit works this way

timoschilling, 2021-04-07:
That can work with this firmware you will garage door will always show as open or as closed (depending on the close sensor config). You need a close sensor, otherwise you never know that the door is close, you can only expect it.

The difference between the window cover and the garage door mode is that in wc mode the device can detect when the motor stops moving and using the move time to verify the state. That no possible with garage doors because the motor works in a different way.

RobbyesHomekit, 2021-04-07:
I'm not an expert. In my case, I have this kind of garage door and I think it works just like the roller shutter.
https://youtu.be/Tnyzg8d7DuY

timoschilling, 2021-04-07:
does you run it currently with the homekit shelly in garage door mode or in window cover mode?

RobbyesHomekit, 2021-04-07:
[image]

garage door mode.. saracinesca box

timoschilling, 2021-04-07:
then it works different as the window cover, if it works the same as a window cover it wouldn't work with garage door opener.

RobbyesHomekit, 2021-04-07:
I think I'll have to keep the shelly 2.5 with the current firmware 😅

timoschilling, 2021-04-07:
What kind of firmware is it?

rojer, 2021-04-07:
i'm confused as to what is being requested here... garage door opener mode needs a "fully closed" sensor and a (rough) value for the time it takes to open the door. seems simple enough.

timoschilling, 2021-04-07:
It sound like he has a different homekit firmware on it and want to switch to our. But I would agree with you a close sensor is needed otherwise you never know it's closed or not.

RobbyesHomekit, 2021-04-07:
This https://drive.google.com/drive/mobile/folders/116XhatLj9XFlAy2zn7QG0U6CXE2ozhu8?usp=sharing

file is master2mb.bin
In file readme 17) shelly25 shelly25 garage door

timoschilling, 2021-04-07:
Ok, now it makes senesce what you ask, but from my opinion the answer is: Yes it is possible to build an garage door without close sensor, but I don't recommend it, because you never know it's the door closed or not. You will leave the home and you garage stays open, not nice.

RobbyesHomekit, 2021-04-07:
I understand what you say but it never happened in two years. In my case the sensor is quite uncomfortable to position

rojer, 2021-04-07:
close sensor is required. it both makes code simpler (by giving us an "anchor point" and not having to track state changes as closely) and improves actual physical security. i suggest you install the sensor, for your own safety, as in this case, as they say, "past record is not an indicator of future performance".

closed 2021-04-07 by rojer