mongoose-os-apps/shelly-homekit #617: Testing 2.9.0

rojer, 2021-04-08:
Big highlight of the release is support of RGBW2 by @timoschilling , @jobl1978 and @andyblac (kudos!)

http://rojer.me/files/shelly/2.9.0-alpha1/

rojer, 2021-04-08:
#615

gregDEweb, 2021-04-09:
Hi many thanks for all the effort you guys are getting into it. 
Would like to try it on a rgbw2 but can‘t get it to load via OTA upgrade. 

http://x.x.x.x/ota?url=http://rojer.me/files/shelly/2.9.0-alpha1/shelly-homekit-ShellyRGBW2.zip

Any ideas whats going wrong on my side?

andyblac, 2021-04-09:
> Hi many thanks for all the effort you guys are getting into it.
> Would like to try it on a rgbw2 but can‘t get it to load via OTA upgrade.
> 
> http://x.x.x.x/ota?url=http://rojer.me/files/shelly/2.9.0-alpha1/shelly-homekit-ShellyRGBW2.zip
> 
> Any ideas whats going wrong on my side?

should be no reason, are you in color mode ? (not base), ie try chaning to white and back to color then retry ota

gregDEweb, 2021-04-09:
Did not help. I updated to the latest version now and started working after a reset

Cojaeck, 2021-04-10:
Great work. Thank you guys.
But I have a strange behaviour since. Just five LEDs of the hole strip show different colours. To make clear, that this is not an hardware issue on my side (just another bloody coincidence;-)) I tried to revert to stock. But the button does not work. I rebooted the device a few times, removed it from power and so on. Any ideas?

andyblac, 2021-04-10:
> Great work. Thank you guys.
> But I have a strange behaviour since. Just five LEDs of the hole strip show different colours. To make clear, that this is not an hardware issue on my side (just another [...]

try this file `http://rojer.me/files/shelly/stock/SHRGBW2-color.zip`

Cojaeck, 2021-04-10:
Thanks. That worked. Now I know that five white LEDs of my strip are dead. As expected no problem with your firmware. I will go back to the Alpha.

andyblac, 2021-04-10:
> Thanks. That worked. Now I know that five white LEDs of my strip are dead. As expected no problem with your firmware. I will go back to the Alpha.

@rojer it seems revert to stock is broken in alpha1, are we reverting to `SHRGBW2` or `SHRGBW2-color` ?

rojer, 2021-04-10:
@andyblac i updated STOCK_FW_MODEL, will be in alpha2

Cojaeck, 2021-04-11:
The actual version does not support 4 channel mode (white), correct? Will that be possible soon?

andyblac, 2021-04-11:
> The actual version does not support 4 channel mode (white), correct? Will that be possible soon?

on the list but will be the last mode to do, iirc.

rojer, 2021-04-12:
pushed alpha2

pivale, 2021-04-16:
Been using alpha2 for almost 4 days, in 1pm, 2.5, plug-s and i3
all working great so far 👍

andyblac, 2021-04-16:
i occasionally get "no responce" from 1 or 2 devices, but exiting home app, and relaunching goes away. not sure if it homekit, home app, or something else.

timoschilling, 2021-04-16:
@rojer do not release yet, maybe there is a issue with detached config. Will investigate later and if yes I will provide details

JKSProds, 2021-04-17:
By the way, what about adaptive lighting in Shelly rgb? Is it on your plans?

timoschilling, 2021-04-17:
@JKSProds Feel free to open a separate issue as feature request for that.

jobl1978, 2021-04-18:
@JKSProds Apple did not update their "non-commercial hap protocol" specification since July 2019. It's currently not clear if adaptive light will be available for devices which are not in the MFI program (HomeKit certificated).

jobl1978, 2021-04-18:
Otherwise the RGBW2 does not know for example what 2000K temperature will be on the currently connected light strip. Every light strip has its own characteristic which RGBW2 doesn't know. HUE or EVE can do that, because they know exactly the specification of their light sources. In short words from myside, its a senseless feature for RGBW2 because it would not achieve the goal why it was invented to match the ambient temperature.

rojer, 2021-04-22:
how are we doing on RGBW2 testing? ok to cut 2.9 or do you intend to make more changes?

timoschilling, 2021-04-22:
I have one last thing which I try to fix now, it's that the lights goes out during a update and hk settings change.
We have one problem with noisy inputs when the shelly is switched with 230v and the switch cable (https://github.com/mongoose-os-apps/shelly-homekit/issues/72#issuecomment-824637481), for now I would say we don't support and on the long term we can maybe write a noisy input which don't need the hardware timer or find a way to use pwm and noisy input with the same timer.

rojer, 2021-04-22:
sounds good. yes, we should find a way to share the timer, perhaps by creating a "timer multiplexer" shim.
i'm ok with it not getting into 2.9

timoschilling, 2021-04-22:
one other bug with the change from rgb to rgbw mode, @jobl1978 take a look on that

timoschilling, 2021-04-22:
@rojer after the merge of #644, for me it's ready to release

Cojaeck, 2021-04-23:
Mine works fine with no issues so far.

Regards
Michael

Von unterwegs gesendet

rojer, 2021-04-24:
pushed alpha3 with working password protection. UI and UX is ready, so please help test.
the only remaining part is protecting HTTP endpoint access (other than root).

rojer, 2021-04-24:
@andyblac for you in particular: please test interaction with the script. i left Shelly.GetInfo open and accessible without authentication, it contains minimum information that should be sufficient for the script. in particular, i added `auth_en` key to reflect whether authentication is required or not.

```
$ curl http://shelly1-test.local/rpc/Shelly.GetInfo
{"device_id": "shelly1-68B743", "name": "Shelly1-test", "app": "switch1", "model": "Shelly1", "stock_fw_model": "SHSW-1", "host": "Shelly1-test.local", "version": "2.9.0-alpha3", "fw_build": "20210424-045846/2.8.2-33-g0193897-master", "uptime": 15, "failsafe_mode": false, "auth_en": true}
$ curl http://shelly1-test.local/rpc/Shelly.GetInfoExt
$ curl --digest -u admin:1234 http://shelly1-test.local/rpc/Shelly.GetInfoExt
{"device_id": "shelly1-68B743", "name": "Shelly1-test", "app": "switch1", "model": "Shelly1", "stock_fw_model": "SHSW-1", "host": "Shelly1-test.local", "version": "2.9.0-alpha3", "fw_build": "20210424-045846/2.8.2-33-g0193897-master", "uptime": 21, "failsafe_mode": false, "auth_en": true, ...}]}
```

andyblac, 2021-04-24:
> @andyblac for you in particular: please test interaction with the script. i left Shelly.GetInfo open and accessible without authentication, it contains minimum information that should be sufficient [...]

it seems flashing is also unprotected atm ?
```
[Write Flash]
Device: 192.168.3.68 is Online
Device URL: http://192.168.3.68/rpc/Shelly.GetInfo
get_uptime: 52
revert False
Downloading Firmware...
Remote Download URL: http://rojer.me/files/shelly/2.9.0-alpha2/shelly-homekit-ShellyRGBW2.zip
Now Flashing HomeKit 2.9.0-alpha2
requests.post('http://192.168.3.68/update', files=files
1 Update applied, rebooting
```

rojer, 2021-04-24:
> > @andyblac for you in particular: please test interaction with the script. i left Shelly.GetInfo open and accessible without authentication, it contains minimum information that should be sufficient for the script. in particular, i added `auth_en` key to reflect whether authentication is required or not.
> > ```
> > $ curl http://shelly1-test.local/rpc/Shelly.GetInfo
> > {"device_id": "shelly1-68B743", "name": "Shelly1-test", "app": "switch1", "model": "Shelly1", "stock_fw_model": "SHSW-1", "host": "Shelly1-test.local", "version": "2.9.0-alpha3", "fw_build": "20210424-045846/2.8.2-33-g0193897-master", "uptime": 15, "failsafe_mode": false, "auth_en": true}
> > $ curl http://shelly1-test.local/rpc/Shelly.GetInfoExt
> > $ curl --digest -u admin:1234 http://shelly1-test.local/rpc/Shelly.GetInfoExt
> > {"device_id": "shelly1-68B743", "name": "Shelly1-test", "app": "switch1", "model": "Shelly1", "stock_fw_model": "SHSW-1", "host": "Shelly1-test.local", "version": "2.9.0-alpha3", "fw_build": "20210424-045846/2.8.2-33-g0193897-master", "uptime": 21, "failsafe_mode": false, "auth_en": true, ...}]}
> > ```
> 
> it seems flashing is also unprotected atm ?, this was flashing alpha3 with passwod set, back to alpha2, no script changes
> 
> ```
> [Write Flash]
> Device: 192.168.3.68 is Online
> Device URL: http://192.168.3.68/rpc/Shelly.GetInfo
> get_uptime: 52
> revert False
> Downloading Firmware...
> Remote Download URL: http://rojer.me/files/shelly/2.9.0-alpha2/shelly-homekit-ShellyRGBW2.zip
> Now Flashing HomeKit 2.9.0-alpha2
> requests.post('http://192.168.3.68/update', files=files
> 1 Update applied, rebooting
> ```

yes, this will be in part 2 - non-rpc http endpoints will be covered with the same digest auth.

andyblac, 2021-04-24:
> yes, this will be in part 2 - non-rpc http endpoints will be covered with the same digest auth.

ok then atm nothing forme to do / test with.

andyblac, 2021-04-24:
ok i understand, i can work on things like hap code etc, but it is no responce for error
```
Host: http://shellyrgbw2-FD89C2.local
Device Name: shellyrgbw2-FD89C2
Model: ShellyRGBW2
Device ID: shellyrgbw2-FD89C2
SSID: None
IP: 192.168.3.68
RSSI: None
Up Time: 0:05:20
Firmware: HomeKit 2.9.0-alpha3 ✔
Configuring HomeKit setup code...
requests.post(url='http://192.168.3.68/rpc/HAP.Setup', json={'code': '111-22-333'}

```

imo would be nice for an error reported.

rojer, 2021-04-24:
wdym? there's a 401 response sent.

andyblac, 2021-04-24:
> wdym? there's a 401 response sent.

requests is returning nothing look,
```
    logger.debug(f"requests.post(url='http://{wifi_ip}/rpc/HAP.Setup', json={value})")
    response = requests.post(url=f'http://{wifi_ip}/rpc/HAP.Setup', json={'code': self.hap_setup_code})
    logger.trace(response.text)
```

```
Configuring HomeKit setup code...
requests.post(url='http://192.168.3.68/rpc/HAP.Setup', json={'code': '111-22-333'}

Devices found: 1 Upgradeable: 0
```

rojer, 2021-04-24:
probably gets raised as an exception.

andyblac, 2021-04-24:
> probably gets raised as an exception.

ok yes if i use status code i get 401
```
Configuring HomeKit setup code...
requests.post(url='http://192.168.3.68/rpc/HAP.Setup', json={'code': '111-22-333'})

401
```

andyblac, 2021-04-24:
do you have any preferance how to handle user/pass in script ? 
i.e
`./flash-shelly.py password ****** shellyrgbw2-FD89C2.local`

rojer, 2021-04-24:
let's mimick curl: --user=user:pass

andyblac, 2021-04-24:
> let's mimick curl: --user=user:pass

ok and for scan mode, if all devices have a differant password ?

andyblac, 2021-04-24:
> > let's mimick curl: --user=user:pass
> 
> ok and for scan mode, if all devices have a differant password ?

prompt user to enter ?

rojer, 2021-04-24:
won't be supported for now, but this is where we might consider introducing a config. to begin with, let's assume the same password for devices thet require it.

oh, and there's only "admin" user now, no need to set it explicitly, so we can just have "--password=pass"

andyblac, 2021-04-24:
> won't be supported for now, but this is where we might consider introducing a config. to begin with, let's assume the same password for devices thet require it.
> 
> oh, and there's only "admin" user now, no need to set it explicitly, so we can just have "--password=pass"

ok

andyblac, 2021-04-24:
```
Configuring HomeKit setup code...
security: True
requests.post(url='http://192.168.3.68/rpc/HAP.Setup', auth=requests.auth.HTTPBasicAuth('admin', 'zxcvbn'), json={'code': '111-22-333'})

401
```

does not seem to be working ?

rojer, 2021-04-24:
has to be digest auth, not basic.

andyblac, 2021-04-24:
> has to be digest auth, not basic.

ok thanks

andyblac, 2021-04-24:
> has to be digest auth, not basic.

working now, 
```
Configuring HomeKit setup code...
security: True
requests.post(url='http://192.168.3.68/rpc/HAP.Setup', auth=requests.auth.HTTPDigestAuth('admin', 'zxcvbn'), json={'code': '111-22-333'})
null

200
```

andyblac, 2021-04-24:
@rojer add support to https://github.com/mongoose-os-apps/shelly-homekit/pull/637 please give it a test, it should support both stock and homekit firmwares.

pivale, 2021-04-24:
> @rojer <https://github.com/rojer> added support to #637

Is there a way to change the password or recover if you forget it? Maybe
not having a pass (or a default) in AP mode

andyblac, 2021-04-24:
> Is there a way to change the password or recover if you forget it? Maybe not having a pass (or a default) in AP mode
> -- Pedro Vale

also with no confirmation of password it could lead to ui lockouts.

rojer, 2021-04-24:
wifi reset also resets password

andyblac, 2021-04-24:
@rojer just added suppport in script for netrc in pull request 

add support for netrc, place a file called ',netrc' in the tools folder containing machines and usernames, passwords

example:
```
machine shellygas-50029178B781.local
login admin
password zxcvbn

machine shellyrgbw2-fd89c2.local
login admin
password asdfgh
```

timoschilling, 2021-04-24:
It should be possible to place the .netrc file in the user home dir as well, then curl will use the file too

andyblac, 2021-04-24:
> It should be possible to place the .netrc file in the user home dir as well, then curl will use the file too

problem with that getting home locations in all OS's (windows, etc), i'm not sure os library has a function,and `~` will not work

rojer, 2021-04-24:
first time i hear about netrc. sure, it can store credentials, but there is no standard function to write the file. i like JSON better as aofrmat for this. 
and i agree with @timoschilling - this belongs in user's home directory, at least on linux and mac os. i'm sure windows has folders for this too (appdata?). it's not urgent, we have the time to figure it out and do it right.
my suggestion: let's not rush this and leave config/credential storage for later. please remove it from current PR.

andyblac, 2021-04-24:
> first time i hear about netrc. sure, it can store credentials, but there is no standard function to write the file. i like JSON better as aofrmat for this.
> and i agree with @timoschilling - this [...]

ok i'll create a fork on my git, then remove save me redoing incase we can use parts.

andyblac, 2021-04-24:
@rojer ok removed .netrc support for now.

timoschilling, 2021-04-24:
$HOME\_netrc on Windows

andyblac, 2021-04-24:
> `$HOME\_netrc` on windows

thanks, i'll make a note, but i think @rojer is leaning towards a json file.

andyblac, 2021-04-24:
> first time i hear about netrc. sure, it can store credentials, but there is no standard function to write the file. i like JSON better as aofrmat for this.
> and i agree with @timoschilling - this [...]

just for info, i added netrc so we could support security on stock as everything is locked down, even GetInfo, so with current code (netrc removed) stock scan mode will not show info of devices, unless all devices are using same username / password.

johannes4576, 2021-04-24:
Working fine by me. I bought an RGBW2 and used it to an already mounted RGB Strip. 

Works fine! Thx! :)

rojer, 2021-04-26:
ok, alpha4 is out with [RGBW2 state fix](https://github.com/mongoose-os-apps/shelly-homekit/pull/649) and final part of the password protection change - this is now fully done.

[25 more replies]

closed 2021-04-30 by rojer