mongoose-os-apps/shelly-homekit #624: OSError: [WinError 10049] The requested address is not valid in its context

ericbrian, 2021-04-09:
I tried to run the flash-shelly.py and got an error:

PS C:\Users\Eric\Documents\GitHub\shelly-homekit\tools> python .\flash-shelly.py -la
Scanning for Shelly devices...

flash-shelly version: 2.6.3
Try to update your script, maybe the bug is already fixed!
Traceback (most recent call last):
  File "C:\Users\Eric\Documents\GitHub\shelly-homekit\tools\flash-shelly.py", line 1199, in <module>
    main.device_scan()
  File "C:\Users\Eric\Documents\GitHub\shelly-homekit\tools\flash-shelly.py", line 1048, in device_scan
    self.zc = zeroconf.Zeroconf()
  File "C:\python\lib\site-packages\zeroconf\__init__.py", line 2508, in __init__
    self._listen_socket, self._respond_sockets = create_sockets(
  File "C:\python\lib\site-packages\zeroconf\__init__.py", line 2343, in create_sockets
    respond_socket = new_respond_socket(i, apple_p2p=apple_p2p)
  File "C:\python\lib\site-packages\zeroconf\__init__.py", line 2302, in new_respond_socket
    respond_socket = new_socket(
  File "C:\python\lib\site-packages\zeroconf\__init__.py", line 2250, in new_socket
    s.bind((bind_addr[0], port, *bind_addr[1:]))
OSError: [WinError 10049] The requested address is not valid in its context

Devices found: 0 Upgradeable: 0

andyblac, 2021-04-09:
zeroconf is failing to detect your lan address for scan, do you have multiple ethernet address ?, or is your pc on a external ip ?

ericbrian, 2021-04-09:
> 
> 
> zeroconf is failing to detect your lan address for scan, do you have multiple ethernet address ?, or is your pc on a external ip ?

It is my laptop on Wifi.

andyblac, 2021-04-10:
as i dont have a windows system here, and it seems to be a python library issue not sure what i can do.

ericbrian, 2021-04-13:
I bought a mac mini and it works..... not without errors, but it works. It found 2 vintage bulbs that I guess you don't support with Homekit. But it did not find the environment sensor. 

Thanks for the help, though. I appreciate it. :)

closed 2021-04-13 by ericbrian

timoschilling, 2021-04-14:
@ericbrian spoiler: support for the vintage will come in the next version. The problems with all Shelly sensors is that they going to a deep sleep to safe battery, that’s why the script can’t find them.