mongoose-os-apps/shelly-homekit #627: rename device

lucaspinelli85, 2021-04-11:
Is possible to rename the device to make more discoverable in the wireless client id in my airport?
[image]
[image]

andyblac, 2021-04-11:
i take it they are all brand new devices, that have not been setup yet ?

andyblac, 2021-04-11:
if theyhave been setup and connected to your wifi,you can cnage the name in General settings
https://github.com/mongoose-os-apps/shelly-homekit/wiki/General-Settings

lucaspinelli85, 2021-04-11:
i already changed the name, but the one displayed is the id
[image]
[image]
[image]

lucaspinelli85, 2021-04-11:
So the real question is: is it possible to rename the device id?

andyblac, 2021-04-11:
> So the real question is: is it possible to rename the device id?

yes, device id is hardcoded to the device, the option above changes the hostname of the device.

andyblac, 2021-04-11:
@rojer is there another dns option that needs tweaking to use hostname ?

timoschilling, 2021-04-11:
> yes, device id is hardcoded to the device

That's not true, it is changeable via the config key `device.id`, but maybe that can have side effects. @andyblac what you mean is the chip id and the chip id is part of the device id.

@lucaspinelli85 there is somewhere a bug in the dns setup because of that routes are showing the device id and not the hostname.

I can't find the issue, but if I remind it right there is one place in the dns setup where we use the device id and not the hostname.

andyblac, 2021-04-11:
> That's not true, it is changeable via the config key `device.id`, but maybe that can have side effects. @andyblac what you mean is the chip id and the chip id is part of the device id. [...]

i meant not user changable ;)

lucaspinelli85, 2021-04-11:
For example, the iPhone, the MacBook, and the HomePod are showed by his hostname! Only the shelly are showed by device id and not hostname in wireless client list and make difficult identify them

timoschilling, 2021-04-11:
Changing the device id is the wrong way, mongoose need to fix the dhcp request. See last line of my previous comment. 

@rojer can you fix that or give me a hint?

lucaspinelli85, 2021-04-11:
so to see the host name instead of the device id in the wireless client list I have to wait for a fix on a next update? correct?

andyblac, 2021-04-11:
> so to see the host name instead of the device id in the wireless client list I have to wait for a fix on a next update? correct?

correct, wait for a fix in mongoose.

lucaspinelli85, 2021-04-12:
Ok i'll wait it!

zhoucky, 2021-04-17:
Hi, my Linksys router is showing the hostnames in the device list. In the summerhouse I have D-link, it’s also showing the hostnames.

lucaspinelli85, 2022-01-05:
I keep seeing the device-id instead of the device-name

rojer, 2022-01-05:
we use names that are guaranteed to be unique until name conflic resolution is implemented.
see https://github.com/mongoose-os-apps/shelly-homekit/commit/acb57e02caef82438a2c82da5c67265a8c76fda0

lucaspinelli85, 2022-01-05:
> we use names that are guaranteed to be unique until name conflic resolution is implemented.
> 
> see https://github.com/mongoose-os-apps/shelly-homekit/commit/acb57e02caef82438a2c82da5c67265a8c76fda0

ok, it's not an urgent serious problem! Better to guarantee uniqueness!