mongoose-os-apps/shelly-homekit #629: Auto reboot

Marcusyo, 2021-04-13:
Every now and then my shellys have no connection to homekit but i can reach then over the web interface. When i reboot them they work just fine. Could you inplement an option for an auto reboot like every day at xx o clock or once a week at xx o clock. I have this problem a lot with my shelly 1 garage opener when im comming home from work and cant open my garage via carplay cause the shelly has no connection to homekit. Iphone is already connected to wifi at this point. Got an homepod mini as my main homekit point.

timoschilling, 2021-04-13:
Your problem sounds like an issue which is fixed in 2.9.0-alpha2, please try that version.

andyblac, 2021-04-13:
sounds like duplicate issue https://github.com/mongoose-os-apps/shelly-homekit/issues/129

closed 2021-04-13 by andyblac

rojer, 2021-04-13:
yes, if "a while" is 25 days, then this is it.

rojer, 2021-04-13:
no need to try alpha, fix will be in 2.9.0, just wait for that and update.