mongoose-os-apps/shelly-homekit #631: Can't seem to update shelly2.5 OTA

peterpeter9999, 2021-04-15:
Hey guys, I'm not able to get the OTA firmware install to work. Do I need to factory reset the shelly and then connect to the shelly wifi? I tried that as well. The update shows like it is starting but it never updates. Please help!

timoschilling, 2021-04-15:
1. Connect the shelly to your wifi
2. Update to the latest stock version
3. open http://A.B.C.D/ota?url=http://rojer.me/files/shelly/shelly-homekit-Shelly25.zip

peterpeter9999, 2021-04-15:
1. done
2. done
3. When navigating to http://192.168.33.1/ota?url=http://rojer.me/files/shelly/shelly-homekit-Shelly25.zip it never loads.

peterpeter9999, 2021-04-15:
[image]

AnakinSpain, 2021-04-15:
Check ip Shelly and  change it

timoschilling, 2021-04-15:
192.168.33.1 is normally only the IP in the AccessPoint mode, I would expect that it has a different IP in your Wifi, take a look on your router maybe it has a host list where you can find the IP.

peterpeter9999, 2021-04-16:
Ok, found the IP address. 

Shows updating but never updates? 
[image]

peterpeter9999, 2021-04-16:
That was it! Found the IP using network analyzer app. Thanks guys!

closed 2021-04-16 by peterpeter9999