mongoose-os-apps/shelly-homekit #632: Move shelly to another network

Ronneke, 2021-04-16:
Hi, I’ve successfully added my shelly devices to my home network. Now I want to move one of them to my son’s house. There the Shelly will not find my network, so how can I change the ssid/password when it’s in is new environment?

timoschilling, 2021-04-16:
There are  ways depending where your Shelly is right now. 

If it’s still in you old WiFi: open the web ui and disable the WiFi, the Shelly then will be accessible in the access point mode and you can move to the new location and configure them there for the new WiFi . 

If you’re already on the new location: use the reset function to clear all config settings. After that your Shelly is in the access point mode. 
How you can reset your Shelly depends on the model you have.

Ronneke, 2021-04-17:
Thx, that did it. (Switching off WiFi on the shelly reverted it to AP mode. Then I could change network).

closed 2021-04-17 by Ronneke