mongoose-os-apps/shelly-homekit #633: Updating time in Home app

brpe001, 2021-04-17:
When opening Apple Home app, it takes 5 to 10 seconds to update ALL installed devices and to be blue to execute actions. I thought this was an Apple issue but then realized this was due to Shelly devices with HomeKit firmware. I have removed them and everything back to normal with Home app. It this lengthy connection time issue something that could be resolved? Or keep them connected continuously somehow? Thanks.

timoschilling, 2021-04-17:
Please provide some more details:
1. Does it happen in one room or all room or on the home overview page?
2. How many Shellys do you have?
3. How may other devices do you have and what kind of?
4. Do you maybe have a shelly in your home which is unconnected to power? For example an unused Plug2?

rojer, 2021-04-17:
also, please open debug log on one of the devices, keep the tab open and then open the home app to capture the moment delay happens.

sagistn, 2021-04-18:
Same here, after 10 seconds of updating, all Shelly are marked as Not Responding.
To @timoschilling 's questions:
1. It happens on all devices (even non-Shelly)
2. 20-25 Shellies installed
3. 2 other devices connected via Homebridge
4. I actually have 1 Shelly 2.5 that is not responding constantly

As for the log, of course now then I have to replicate everything works perfectly.
Will update as soon as I have it.

timoschilling, 2021-04-18:
@sagistn remove your not responding 2.5, the problem is that if Home fetches the updates from the accessories it does it in synchron way, one device after each other and it will wait for devices which aren't responding, so your 2.5 will block the updating of all others.

sagistn, 2021-04-18:
Makes sense. I'll try that.

brpe001, 2021-04-18:
Thanks for quick answers. This happens when opening Home app, no difference whether it is the main screen or specific room. I have around 50 different Philips Hue lamps / bulbs all connected through the Philips Hue bridge to HomeKit. No other devices except 3 Shelly 1 I recently installed. At the beginning I thought the issue was on HomeKit but after a while I started thinking it was because the Shelly devices. I have removed them and now the issue does not happen.  My experience not being a techie person is that when opening HomeKit the time it takes for HomeKit to get the status of the Shelly devices is too long resulting in this issue. Once this issue has happened and after a while seems HomeKit refreshes again the status and everything ok. This is anyhow annoying as for instance this also happens when asking Siri via HomePod to switch on a light, doesn’t matter if Shelly or Hue, the response from Siri is “keep trying” and then just “not possible”.

timoschilling, 2021-04-18:
@brpe001 there is a difference between bridged accessories and non bridged accessories, all your 50 difference bridge accessories are menaged with on request to the bridge. Some questions about your shellys: 1. Can you post the RSSI value of the Wifi section and Connections of the HomeKit section of all your shellys. 2. Did it help for a time if you reboot your shellys? Please try it.

brpe001, 2021-04-18:
I just reinstalled one of them back to HomeKit. The RSSI is -44 and the HomeKit connections are 0 pending 12 active, 12 max. So far with one seems to work fine. Let’s see how it evolves and I will then connect another one and then the third one.

brpe001, 2021-04-18:
I just reinstalled one of them. The RSSI is -44 and the HomeKit connections are 0 pending 12 active, 12 max.

brpe001, 2021-04-21:
I have reinstall the three Shelly as HomeKit enable now. Let’s see the behavior. I am starting to think this is more related to the Phillips Hue Bridge as I noticed that when the issue occurs there are 2 or 3 bulbs also not reachable from Hue app itself. Need to investigate in that direction. I have 43 bulbs and 8 accessories and this still under the recommended limit so not sure why the Hue bridge now having this issue. Anyhow thanks for taking the time to respond to initial request.

andyblac, 2021-04-21:
> I have reinstall the three Shelly as HomeKit enable now. Let’s see the behavior. I am starting to think this is more related to the Phillips Hue Bridge as I noticed that when the issue occurs there [...]

i had nothing but issues with my homekit, once my hue got over 30 accessories in the end i ditched hue, i got up too 116 device's over 2 hubs, since i ditched it hk it alot better, not perfect by no means but loads better.

closed 2021-06-02 by github-actions[bot]