mongoose-os-apps/shelly-homekit #635: Shelly 1 - Garage Door Notification

Ludelag, 2021-04-19:
Hello, I just notice that since the last update 2.8.2 when is asked to Siri to open the garage Door, doesn't say any more opening garage, now say garage open, or asking close garage the answer from Siri is not any more closing garage, Now the message is garage closed, when the action is still taking place. 

I use Shelly as normally close, also I notice that when you ask to open, the status showed in the Homekit is constantly opening and does not indicate anymore after the seconds programmed the open status. 

There is any change in the last update regarding this situation??

Thank you

timoschilling, 2021-04-22:
Do you know which version do you had before?

Ludelag, 2021-04-23:
I guess was the previous one, I use to update when the new version released

timoschilling, 2021-04-23:
I can't find any related changes between 2.8.0 and 2.8.2, but between 2.7.3 and 2.8.0 are some possibilities.
Can you please revert to [2.7.3](https://github.com/mongoose-os-apps/shelly-homekit/releases/tag/2.7.3) and test it there.

timoschilling, 2021-04-23:
and can you please read #535, sound like the same issue or?

Ludelag, 2021-04-26:
Dear Timo, today in the afternoon I will revert and let you know what I find.

angyp, 2021-04-30:
Hey Guys, Based on this thread when I implemented this for the first time I decided to use 2.7.3 instead of the latest. I found the same thing as @Ludelag did with the Siri responses. Is it worth trying an older version? Or is it an IOS version problem. I am using the latest 14.5.

Ludelag, 2021-04-30:
I revert already to the version 2.7.3 as @timoschilling advice, but is the same thing. I guess this has to do with IOS version I have the new version now. Now in Home kit shows closing but Siri still says is close or is open.

timoschilling, 2021-04-30:
Ok if the status in Home is right but not in Siri, it’s a Siri bug. This firmware has no knowledge about Siri it only specks to HomeKit. 

What’s you Siri language? English?

Ludelag, 2021-04-30:
Spanish. Maybe this is the reason.

timoschilling, 2021-04-30:
Could be a translation issue inside of Siri.

andyblac, 2021-04-30:
@Ludelag ok to close ?

Ludelag, 2021-04-30:
@andyblac for me yes. Thank you for the attention.

closed 2021-04-30 by andyblac