mongoose-os-apps/shelly-homekit #639: roller shutter with legrand valena life push button

zsolt-dev, 2021-04-21:
I bought this roller shutter buttons from Legrand, since IMO they have the best looking design:
[image]

They have 3 buttons: up, down and stop (the button in the middle).
It connect to just two cables (up&down).

How does the stop button works:
1. If you press the down button, then the stop button, it sends a signal to the UP wire
2. if you press up and then the stop button, it sends the signal to the DOWN wire.

So it basically remembers the last choice. It is probably mechanical, since I do not think there is any electronics inside.

So I basically need the shelly to stop the rollers when I press the opposite direction button, instead of the same direction button.

The question is:
Can we have one more mode, called "seperate - momentary (reverse stop)" ?

Thank you

link to the roller button:
https://www.legrand.com/ecatalogue/752230-roller-blind-push-button-valena.html

louis49, 2021-04-22:
I've the same problem with roller shutter buttons from Schneider Electric that have a better looking design   https://www.se.com/ww/en/product/S520208/odace%2C-rocker-switch%2C-3-positions%2C-6-ax%2C-white/

[image]

zsolt-dev, 2021-04-22:
I did not see those anywhere... They look great. 

The existence of these validate the need for the reverse stop mode.

jobl1978, 2021-04-22:
I think having a third button (stop) for window covering is a design fail. There is no need to have a dedicated button for stop. Normally window cover switches work in that way if you press down/up window cover moves down/up. If window cover is already in move an additional press to up/down stops it. The actual Shelly HK implementation only stops if you press in the same direction as the motor currently moves.

This one is window covering switch from Berker
[image]

zsolt-dev, 2021-04-22:
> The actual Shelly HK implementation only stops if you press in the same direction as the motor currently moves. In my opinion the HK firmware should be changed that pressing a button should always stop a already moving window cover before changing the move direction.

I agree, this should be the solution.

jobl1978, 2021-04-22:
@zsolt-dev I did the code change for the Shelly25 HK firmware. Its currently pushed to the server and is waiting for acceptance ;-)

zsolt-dev, 2021-04-23:
@jobl1978 Thank you Josef! I did not expect this to be resolved so fast. You are what makes this community great. Thank you again

timoschilling, 2021-04-24:
@rojer @andyblac can be closed now after the merge

closed 2021-04-24 by andyblac

zsolt-dev, 2021-05-03:
I just wanted to report to someone looking at this in the future, to NOT buy the legrand valena life push button.

As it turns out, the stop button presses the up&down at the same time (or very fast one after another), so it still does not work with this patch.

I have read the datasheet and there is nothing that says what the stop button does:
https://docdif.fr.grpleg.com/general/MEDIAGRP/NP-FT-GT/F02356EN-00.pdf

There are two wiring diagrams, but my button can be connected only as shown on 4.2 (with "Automatisms housing") since the terminals as shown on 4.3 are not accessible.

[image]

@louis49 can you please check how does it work with your Schneider Electric one?

Thank you

jobl1978, 2021-05-03:
Powering up/down at the same time is a totally fail. As I said before, for window covering switches you don't need a stop button. Buy one from Berker and forget that Schneider toy and you will be happy.

https://www.berker.com/de/e-katalog/schalterprogramme-und-gebaeudesteuerung/designlinien/berker-q.3/jalousie-schalter-taster/503520/303144.htm

zsolt-dev, 2021-05-04:
@jobl1978 Thank you for your recommendations.

My whole reason for not using those switches is design. Clicking left to go down and right to go up is not compatible with my OCD :)

I am looking for something that has two buttons, up/down and one is below the other.

If someone is interested what does the valena switches do, when you press DOWN and wait and press STOP.
```
62897021203 shelly_input.cpp:76     Input 2: change (state 1)
62897027710 shelly_hap_window_c:342 WC 1: Tgt pos 100.00 -> 0.00 (ext)
62897033660 shelly_hap_window_c:322 WC 1: State: idle -> move (0 -> 20)
62897042381 shelly_output.cpp:63    Output 2: off -> on (move)
62897048667 shelly_hap_window_c:322 WC 1: State: move -> rampup (20 -> 22)
62897130226 shelly_hap_window_c:544 P = 103.82 -> 124.74
62897136467 shelly_hap_window_c:322 WC 1: State: rampup -> moving (22 -> 23)
62897230695 shelly_hap_window_c:594 Moving to 0, p 128.13
62897283326 shelly_input.cpp:76     Input 2: change (state 0)
62897290982 shelly_hap_window_c:331 WC 1: Cur pos 99.74 -> 99.57, P = 128.01
62897539509 shelly_input.cpp:76     Input 2: single (state 0)
62897930913 shelly_hap_window_c:594 Moving to 0, p 127.94
62898032073 shelly_hap_window_c:331 WC 1: Cur pos 97.68 -> 97.38, P = 127.99
62898721377 shelly_main.cpp:569     Up 62898.71, HAP 0/2/12 ns 2, RAM: 21996/38216; st 59; 4.1: c:1 mp:124.74 mt_ms:33889 cp:95.61 tp:0.00 lemd:2 lhmd:0
62898731202 shelly_hap_window_c:594 Moving to 0, p 127.84
62898832166 shelly_hap_window_c:331 WC 1: Cur pos 95.31 -> 95.02, P = 127.63
62899504113 shelly_input.cpp:76     Input 2: change (state 1)
62899510533 shelly_hap_window_c:594 Moving to 0, p 126.40
62899516206 shelly_hap_window_c:342 WC 1: Tgt pos 0.00 -> 93.02 (ext)
62899523513 shelly_hap_window_c:331 WC 1: Cur pos 93.02 -> 92.98, P = 126.38
62899529672 shelly_hap_window_c:342 WC 1: Tgt pos 93.02 -> 92.98 (fixup)
62899534795 shelly_output.cpp:63    Output 2: on -> off (moving)
62899540897 shelly_hap_window_c:322 WC 1: State: moving -> stop (23 -> 24)
62899547552 shelly_input.cpp:76     Input 1: change (state 1)
62899553743 shelly_hap_window_c:342 WC 1: Tgt pos 92.98 -> 100.00 (ext)
62899578682 mgos_sys_config.c:232   Loading conf2.json
62899719790 mgos_sys_config.c:174   Saved to conf9.json
62899727669 shelly_hap_window_c:322 WC 1: State: stop -> stopping (24 -> 25)
62899743637 shelly_hap_window_c:322 WC 1: State: stopping -> idle (25 -> 0)
62899771968 shelly_input.cpp:76     Input 2: change (state 0)
62899778055 shelly_hap_window_c:322 WC 1: State: idle -> move (0 -> 20)
62899840896 shelly_output.cpp:63    Output 1: off -> on (move)
62899847352 shelly_hap_window_c:322 WC 1: State: move -> rampup (20 -> 22)
62899942394 shelly_hap_window_c:544 P = 111.12 -> 124.74
62899948935 shelly_hap_window_c:322 WC 1: State: rampup -> moving (22 -> 23)
62900046764 shelly_input.cpp:76     Input 2: single (state 0)
62900734228 shelly_input.cpp:76     Input 1: long (state 0)
```

If you press the stop button lightly, it will do only the first click.

So I do not recommend these switches to anyone, since it depends how you press the stop button.

andyblac, 2021-05-04:
> I am looking for something that has two buttons, up/down and one is below the other.

take alook at 3 position retractive switches, thats what we call them here in the uk.

https://www.click4electrics.co.uk/white-modules-c-328_518_707_1072/click-minigrid-md075wh%C2%A0white%C2%A010a-3-position-retractive%C2%A0sw-module-p-5565

zsolt-dev, 2021-05-04:
I have destructively gutted the Valena life for 11Eur just to see what it does.

The stop button presses both buttons at the same time. Before I thought there is some magic, that it remembers the last choice, but its not. It's just a lottery which one will get pressed first, that is why It took me long time to figure out. I am going to open new feature request that work great with switches with stop buttons (legrand, Schneider...) and maybe someone will be willing to code it. Thanks everyone for your responses.

marcus-aa, 2022-06-23:
Hi @zsolt-dev , the reason you have problems with this button is that it is the version made for electronic control devices, relay boxes or the new Somfy DCT motors. Many of them use "both-for-stop" as a way to stop the shutters from one location even if they were started from another. This type of button can NEVER be connected directly to supply current to a motor. 

Here is how it is implemented by Somfy Centralis:
[image]

Here is a summary of the different versions:

https://docdif.fr.grpleg.com/general/MEDIAGRP/NP-FT-GT/F00499EN-06.pdf

There is another version of the buttons, which is made for DIRECT CONTROL of shutters, and that version works perfectly fine with Shelly, if you select toggle input mode. The up or down button are toggled OFF buy the stop button in a mechanical way, and it can never close both contacts.

https://www.legrand.com/ecatalogue/077026-roller-blind-switch-mosaic-for.html

So, there are 3 main versions;

[image]

77025 / 79225 - With STOP, for electronic control, does NOT work with Shelly
77026/ 79226 - With STOP , for direct motor drive, CONFIRMED working with Shelly in toggle mode
77028/79228 - Without STOP, for relay control, SHOULD work with Shelly in momentary mode, (not tested)

So if you are looking for this design, just buy the right one. It will work. I just bought one and have these on all may shutters from before, just starting to add Shellys to automate them. Just like you, I like the stop button an dthe design makes it easy for others to see exactly how it works.

It is also sold in a bag with article number 099656 (without frame), but it says 077026 on the switch inside.

[image]

zsolt-dev, 2022-08-09:
@marcus-aa

Thank you for investigating. Before I replace my switches, can you please try this:

1. close blinds via the wall switch (press only the down button, nothing else)
2. open blinds via the App or siri
3. press the down button on the wall switch (do not press anything else)

Does it close the blinds? Or in other words, does the switch stays pressed down after the rollers finish working?

If yes, this is the reason I bought the version that should be used with "automation" and not the ones for "direct".

I really think this is the correct switch to use for the automatation and the shelly should register the stop (both up and down signals at the same time)