mongoose-os-apps/shelly-homekit #640: Shelly 1L loses connection & operation

wawoodwa, 2021-04-22:
Looking for some advice to troubleshoot this issue. This is for a single 1L I have installed with a LED light and a bypass. I have another same setup (same LED and bypass) and it works great. 

When it fails, it turns completely off. Switch does nothing, no connection to WiFi. Power cycle the breaker, and it all comes back and works correctly. 

What is the best way to troubleshoot this?  Can I configure a syslog server to get what it is doing before it dies? There is no rhyme or reason when it stops.

nicsoellner, 2021-04-22:
I had the exact same problem with my Shelly1 (non L)

I updated to the 2.9 alpha which seems to have fixed it

wawoodwa, 2021-04-22:
I'll give it a try.  Interesting it happens on one but not the other, when they are exact duplicates.

closed 2021-05-30 by github-actions[bot]