mongoose-os-apps/shelly-homekit #641: Shelly 2.5 not flashing properly (?)

nightmechanic, 2021-04-22:
I a have a shelly 2.5  that can't flash the HomeKit fw properly:
I Initially used the flash-shelly.py script, this is what I got:

```
Host: http://shellyswitch25-693239.local
Device Name: None
Model: Shelly25
Firmware: Stock 1.10.3 ➜ HomeKit 2.8.2
Do you wish to flash shellyswitch25-693239 to HomeKit firmware version 2.8.2 (y/n) ? y
Now Flashing...
waiting for shellyswitch25-693239 to reboot....................
still waiting for shellyswitch25-693239 to reboot....................
we'll wait just a little longer for shellyswitch25-693239 to reboot............................................................................................................................................................................................................................................................................................................................................................
Failed to flash shellyswitch25-693239 to 2.8.2
```

It then appeared as an AP in failsafe mode,
so I tried to load the version from a local copy  - same results - failsafe mode,
I tried updating to 2.7.2 form failsafe mode - same , loaded as 2.7.2 in failsafe mode,
I updated to the stock shelly (1.10.3) - that works fine! I get the normal shelly UI, manage to configure everything etc.
every attempt to move to shelly-homekit ends up in failsafe,
reverting to shelly stock seems to work...
help :)
(p.s. I have 7 other shelly 2.5 that migrated to shelly home-kit just fine...)

AnakinSpain, 2021-04-22:
Some 2.5 has a problem with case. Contact with Shelly support

nightmechanic, 2021-04-22:
> Some 2.5 has a problem with case. Contact with Shelly support

Are you referring to the WiFi antenna issue? if so this is not the case here - it works fine with the stock shelly FW...
If not could you please elaborate on the problem?
Thanks!

andyblac, 2021-04-28:
@nightmechanic can you run script with `--log-file debug.log` added to command line and post he debug.log file here i'll take a look

closed 2021-06-06 by github-actions[bot]