mongoose-os-apps/shelly-homekit #645: shelly RGB (2.9.0-alpha3)- initial state:ON - not working after the power is off and back on

Va5a, 2021-04-25:
First, thank you for now supporting Shelly RGB! I'm testing it and it works great. 

One thing: after you turn off the electricity and then turn it back on, shelly "forgets" the last setting (at least brightness is reset to 0%). To test it - set initial state: on, select color /brightness of your choice. then disconnect it from power. When connected again, it won't turn on (at least brightness will be set to 0%). Hope this is useful.

timoschilling, 2021-04-25:
@jobl1978 can you please take a look

jobl1978, 2021-04-26:
@Va5a Did you set "Initial state" to "last"?

timoschilling, 2021-04-26:
@jobl1978 I can reproduce it, Initial state => last works, but Initial state => on not

jobl1978, 2021-04-26:
> @jobl1978 I can reproduce it, Initial state => last works, but Initial state => on not

I will have a look on it

jobl1978, 2021-04-26:
@timoschilling @Va5a Problem found and fixed -> #649

Va5a, 2021-04-26:
Wow. That was fast!! Thank you :)

PR #649 (2021-04-26): Force initial state switching for ShellyRGBW2
Internal caching prevented correct switching of initial state (on/off/input).
fixes #645
closed 2021-04-26 by rojer