mongoose-os-apps/shelly-homekit #646: Shelly Icon Apple HomeKit

Chpedro93, 2021-04-25:
Hi everyone, 
I’m trying to connect a Shelly 1 to an outdoor siren with an automation created with motion sensors.
It’s possible to put the Shelly connect to the siren with a siren icon in the homekit?
Thanks.

timoschilling, 2021-04-25:
Should be possible, I will take a look

timoschilling, 2021-04-26:
The only way to get a siren icon in homekit is to use a "alarm system", but an alarm system isn't mate for your use-case, you can't set the "alarm triggered" state from homekit to the alarm system, only the alarm system can say that it is triggered. A homekit alarm system is mode to represent a alarm system controller in homekit not to represent a siren.

closed 2021-04-26 by andyblac

Chpedro93, 2021-04-26:
Thanks for the answer, so to be able to do it I need a control panel right?

andyblac, 2021-04-26:
add support here https://github.com/mongoose-os-apps/shelly-homekit/issues/552

timoschilling, 2021-04-26:
For your case you should simply use one of the hap types which is supported by the shelly1.