mongoose-os-apps/shelly-homekit #647: Output mode "dual" in garage door

lucaspinelli85, 2021-04-26:
can anyone tell me how it works and what it is for? thank you

timoschilling, 2021-04-26:
https://github.com/mongoose-os-apps/shelly-homekit/wiki/Garage-Door-Opener-Settings

lucaspinelli85, 2021-04-26:
I have read the guide but it does not say what I am looking for!!
[image]

timoschilling, 2021-04-26:
Sorry my fault!

Single is for the wiring you can see in the picture under https://github.com/mongoose-os-apps/shelly-homekit/wiki/Garage-Door-Opener-Settings#shelly-2-or-25 if your garage door has one input for toggle, dual is if your garage door has two inputs, one for closing and one for opening.

lucaspinelli85, 2021-04-26:
ok thanks!

closed 2021-04-26 by andyblac