mongoose-os-apps/shelly-homekit #651: Téléruptor

louis49, 2021-04-27:
Hello! 
Has anyone ever succeeded in connecting a switch plugged to a teleruptor ? Is the connection to the Shelly (2.5) the same? Is there a configuration planned for this ? 

Illustration of this kind of installation : 

[telerupteur.jpg](https://www.astuces-brico.fr/wp-content/uploads/2019/11/telerupteur.jpg)

Thank you

timoschilling, 2021-04-27:
There is no explicit mode for that. But it works already, you simply need to replace the relay with the shelly 2.5.

jsiegenthaler, 2021-04-27:
A telerupter is also called an Eltako 

I do exactly this with my hallway lights, I fitted a Shelly1
Here's the wiring diagram
https://www.shelly-support.eu/forum/index.php?thread/7974-shelly-1-in-einen-mit-n-geschaltete-taster-bzw-eltako-schaltung-integrieren/

timoschilling, 2021-04-27:
Oh yes thanks @jsiegenthaler I missed that wiring option, and both cases works with 1 & 2.5

louis49, 2021-04-27:
@jsiegenthaler Thank you for this information. This diagram requires access either to the electrical panel or to the lamp. There is no way to put a Shelly in parallel like one of the switches?

timoschilling, 2021-04-27:
It’s possible but then you don’t know the status of the lamp, you can only assume the status.

timoschilling, 2021-04-27:
Sorry I missed the description: Use the normal switch mode and set a auto off to 100ms (value depends on the sensitivity of your relay)

closed 2021-06-05 by github-actions[bot]