mongoose-os-apps/shelly-homekit #652: RGBW2 Alpha4

andyblac, 2021-04-27:
@jobl1978 i was removing password and device took ages to reboot, here is a log 
[rgbw2.log.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/6385136/rgbw2.log.zip)

but as you see getting lots of 
```
[Apr 27 11:59:38.237] shelly_output.cpp:74    Output 1: OFF (transition)
[Apr 27 11:59:38.240] shelly_output.cpp:74    Output 2: OFF (transition)
[Apr 27 11:59:38.245] shelly_output.cpp:74    Output 3: OFF (transition)
[Apr 27 11:59:38.249] shelly_output.cpp:74    Output 1: OFF (transition)
```

closed 2021-04-28 by rojer

timoschilling, 2021-04-28:
This should keep open, alpha5 reduced only the log output but didn’t fixed the long boot time

jobl1978, 2021-04-28:
> This should keep open, alpha5 reduced only the log output but didn’t fixed the long boot time

@timoschilling the problem of this issue was a reload hangup after resetting the password. The reduce of the logging was another cleanup which was not the root cause of this issue. The alpha5 rework of @rojer fixed that hangup.

timoschilling, 2021-04-28:
Ok, never mind, I expected a problem that the transition was executed blocking to the rest of boot.