mongoose-os-apps/shelly-homekit #654: Possible Regression: HAP Device type "lock" showing up as Switch in homekit.

arnoldad, 2021-04-29:
Hi, I'm running the following on Shelly 1

Version: 2.8.2
Build: 20210315-093656/2.8.2-gc0f7426-master

The following settings are on the shelly.

[image]

However in homekit, I can only set the device type to Fan, Light, or Switch, but not Lock.

Am I missing something? I was able to get another shelly 1 showing up as a lock a few weeks ago - so i wonder if this may be a regression.

arnoldad, 2021-04-29:
I should add that IOS 14.4.2. And these are the options I get when adding the device.

[image]

Also THANK YOU for your work on this!

andyblac, 2021-04-29:
look further down in UI to "HAP Service Type" change it to "Lock"

https://github.com/mongoose-os-apps/shelly-homekit/wiki/Switch-Settings

arnoldad, 2021-04-29:
@andyblac My screenshot shows HAP Service Type field has value Lock.=) Am I missing anything? Thanks for the docs. link I read it thoroughly.

arnoldad, 2021-04-29:
Update: I needed to reset the homekit server. Previously I had removed the homekit accessory from the Home, this did not help. After using the shelly web interface to reset the homekit server and restarting it (same pairing code) I was able to add the device as a lock type accessory.

closed 2021-04-29 by arnoldad