mongoose-os-apps/shelly-homekit #655: Shelly 2.5 Roller Shutter Status isn’t Right

nyxs, 2021-04-29:
Hey,
I installed the Shelly 2.5 as a roller shutter, and everything works, besides the status.
When the status is set to 50, the roller shutter is actually on around 25.
I performed calibration several times, but it still says the wrong status.

lucaspinelli85, 2021-04-30:
It is correct, not is a problem! 0% is completly closed, 100% is open, but when you're closing, when the roller shutter touch ground it is not 0% but about 20%!

pivale, 2021-04-30:
Hey,

This is a calibration situation that happens because of the type of shutter you are using.
We are already handling this in issue Roller Shutter: positioning accuracy #204

If you want to add your data there, we could use it to make the calibration process better.

@andyblac I think we can close this one, since we are already handling this in issue #204

lucaspinelli85, 2021-04-30:
I instead have another problem that the status in the app don't update if i move roller shutter by botton. Have you any idea of problem?

pivale, 2021-04-30:
@lucaspinelli85 can you check if it's the issue described in on #374?
https://github.com/mongoose-os-apps/shelly-homekit/issues/374#issuecomment-824601873

lucaspinelli85, 2021-04-30:
Maybe similar.
If i open/close by app ok, but if i open/close by toogle, in the app don't change status, it remain costant!

lucaspinelli85, 2021-05-03:
Could you solve this problem? #374

closed 2021-06-11 by github-actions[bot]

lucaspinelli85, 2021-06-30:
I keep having this problem, the shutter status is not updated between app and reality