mongoose-os-apps/shelly-homekit #660: Shelly 2.5 Roller Shutter mode inaktiv

NormanBlu, 2021-05-02:
Hello I have installed version 2.9.0 on my Shelly 2.5. When changing to Roller Shuttler mode, the interface doesn‘t change.H It remains with the settings for the switch. 
You have reported in another post that the manual setting of moving times in the Roller Shutter mode should be implemented in 2.9.0. Has this already happened. In Beta 1, the Shelly has always executed only 1.0xx Sec. steps when calibrating. This problem was already described here several weeks ago.

timoschilling, 2021-05-02:
I noticed a similar problem, change the mode and then restart the Shelly

BlancoDeMallloca, 2021-05-02:
Had the same problems with Shelly 1 switching to garage door mode. After some reboots and changing the mode the UI changed

lucaspinelli85, 2021-05-03:
When programming settings, you have to first set HomeKit code, after you can change mode setting and it works!

closed 2021-06-11 by github-actions[bot]