mongoose-os-apps/shelly-homekit #661: Selly1

cskallo, 2021-05-02:
Hi!
For irrigation control, feedback is only available in lock mode.
Is it possible to have feedback and programming for the time in the irrigation mode. If not, is there a mod for the irrigation icon to be changed in lock mode?

timoschilling, 2021-05-02:
> Is it possible to have feedback and programming for the time in the irrigation mode

Will become in the future, see #263 

> For irrigation control, feedback is only available in lock mode.

can you explain that please

cskallo, 2021-05-02:
Shelly csak lA ock modban küld viszajelzést,jó lenne ha kapni lehetne ötözés modban is

cskallo, 2021-05-02:
Shelly only sends feedback in lA ock mod, it would be nice if you could get alloy in mod as well

timoschilling, 2021-05-02:
I expect that you mean the Lock / Unlock state of you iPhone, right?

message on the iPhone are coming from the Home app and not from the Shelly. Shelly send his state to each connect iOS device, the rest is part of iOS.

cskallo, 2021-05-02:
Hello!
When the selly changes its status from Home application, an notification is received from the status store

timoschilling, 2021-05-02:
what status store?

timoschilling, 2021-05-02:
can you provide screenshots to show us what you expect on what place

andyblac, 2021-05-02:
@cskallo what version firmware you running ? "Water Valve" support only basic support atm, please see here
https://github.com/mongoose-os-apps/shelly-homekit/wiki/Switch-Settings

```
Valve: shown as a water Valve in HomeKit (other options available in 'Valve Type').
Valve Type: Allows you to choose the type of water accessory you would like to be shown in HomeKit (this is only available when HAP Type is set to 'Valve').

Generic Valve: shown as a garden tap in HomeKit.
Irrigation: shown as a Sprinkler Irrigation in HomeKit.
Shower Head: shown as a Bathroom Shower in HomeKit.
Water Faucet: shown as a Kitchen Tap / Faucet in HomeKit. NOTE: Apple's Home app doesn't support the valve in automatisations and scenes, you can work around that by using other app like Home+ or Eve.
```

closed 2021-05-05 by cskallo