mongoose-os-apps/shelly-homekit #663: HomeKit Accessory Security with 2.9.0

RedAnon, 2021-05-02:
Just updated to 2.9.0 on my 1/2.5/PlugS and with HomeKit Accessory Security none of the Shellys work.
Can anyone confirm my issue?

timoschilling, 2021-05-02:
Can you explain "none of the Shellys work", what did you expect? what error you get?

RedAnon, 2021-05-03:
I expected them to work like they did before? AKA no web interface (obviously with "restricted to home") but fully working through the Home apps. AKA just like they are supposed to.

No error. They just never connect. They update for some time and then there is just no response.

timoschilling, 2021-05-03:
Must be an other issue, auth doesn’t effect HomeKit. 

Can you provide logs please

RedAnon, 2021-05-04:
Auth? I'm not talking about the new password protection inside the Shelly firmware. I'm talking about the Apple feature in the new routers (I have the Linksys one) https://support.apple.com/en-us/HT210544 . It worked flawlessly before.

rojer, 2021-05-04:
nothing really changed with respect to HAP or security. can you confirm that it works with 2.8 and not with 2.9?

timoschilling, 2021-05-04:
Please downgrade to 2.8 and retest it with that. There could also be a change in the router or iOS.

RedAnon, 2021-06-02:
Instead of downgrading I've waited and finally updated to 2.9.1. This fixed the issue and everything works now. No idea what the problem was.
Before updating to 2.9.1 I've tested it one last time and with "restricted to home" I was unable to operate any of the accessories.

So this can be closed. :) Hopefully it won't glitch again after some next update. :)

rojer, 2021-06-02:
thanks for the update!

closed 2021-06-02 by rojer