mongoose-os-apps/shelly-homekit #667: shelly 2.5 force opens rollers on reboot

earglass, 2021-05-05:
Hi! I have strange issue. One of 3 of my shelly 2.5 (set as rollers) is rebooting from time to time and opens roller when it does.
I have set device to 'stop' on start and soft reboot on connection loss. However it's not important, as even manual reboot results in roller opening.
All of them work on MQTT protocol and all I'm seeing on moment of reboot is that topic 'shellies/shellyswitch25-lr_rollers_left/roller/0' has message open, when 0/command remains as 'stop'.
Debug log has nothing :(
[log.txt](https://github.com/mongoose-os-apps/shelly-homekit/files/6429956/log.txt)

It becomes problematic as closed roller is opening without me being in home..
Current version of firmware - 20210429-100559/v1.10.4-g3f94cd7

timoschilling, 2021-05-05:
You are using the stock Shelly firmware. This repository is for a different one. Contact Shelly to get support.

earglass, 2021-05-06:
Oh, sorry. I will, thanks.

closed 2021-05-06 by andyblac