mongoose-os-apps/shelly-homekit #671: Shelly RGBW2

stoex, 2021-05-08:
Hi,

today i tried to flash the firmware with this link: http://A.B.C.D/ota?url=http://rojer.me/files/shelly/shelly-homekit-ShellyRGBW2.zip

First of all i connected the Shelly to the wifi and updated the original firmware.

When i tried to flash the firmware with the link above, the shelly didn't do anything. If i go to the ip the original firmware is still installed on it.

Does anyone has the same problem?

timoschilling, 2021-05-08:
You need to set your Shelly in color mode to flash them.

timoschilling, 2021-05-08:
Wenn er im Auslieferungszustand ist er quasi in gar keinem Mode, er muss erst einmal in den Color Mode geschaltet werden, ansonsten lässt er sich mit dieser Firmware nicht flashen.

stoex, 2021-05-09:
> Wenn er im Auslieferungszustand ist er quasi in gar keinem Mode, er muss erst einmal in den Color Mode geschaltet werden, ansonsten lässt er sich mit dieser Firmware nicht flashen. 

Unter Device Type ist ein Haken bei Color gesetzt. Das war standardmäßig bei mir so. Heißt der Modus ist bereits aktiviert oder?

timoschilling, 2021-05-09:
Ist bei Weiß noch ein Hacken? Wenn ja der muss weg

stoex, 2021-05-10:
[image]

timoschilling, 2021-05-11:
I found the problem http://rojer.me/files/shelly/shelly-homekit-ShellyRGBW2.zip produces a 404, please use http://rojer.me/files/shelly/2.9.0/shelly-homekit-ShellyRGBW2.zip

@rojer can you fix that please

timoschilling, 2021-05-11:
@stoex can you please confirm here if it works now

stoex, 2021-05-11:
@timoschilling @rojer thank you! Works great now👍🏻