mongoose-os-apps/shelly-homekit #672: No access shelly 1

Jagigu, 2021-05-09:
I can't access shelly 1, I hear the sound but it doesn't work.

I try to access through the IP and it is not possible either.
Since HomeKit appears as "no response"

What I can do?

Thanks

Ryf81, 2021-05-10:
I had the same issue some days ago. One of my 24 Shelly was not response. I was even not able to use the UI over the device IP.  I try to reboot the Router and cut the power of the Shelly, but it didn't work, so i thought the Shelly is defect. Therefore i ordered a new one. But then, the next day the Shelly was response again, and everything worked as it should. Now i have a Shelly as  replacement device :-)

timoschilling, 2021-05-12:
@Jagigu do you see a wifi shelly1-*****? Maybe something has triggered a reset of your shelly.

iliquifysnacks, 2021-05-18:
I have the same problem. 

The shelly1 is connected to the wifi with good connection quality according to the router:
[image]

I can't ping it though, it only responds with timeouts. I also see no ssid that would indicate the device has reset. 

Interestingly the rgbw2 that I also installed which is still running the original firmware is also only working occasionally. 

Any ideas to resolve this would be highly appreciated :)

kachyy, 2021-05-22:
also with Asus router, all Shellys were online but not accessible

fixed by changing the following in the router **Advanced Settings**:
Wireless --> Professional --> Enable WMM No-Acknowledgement = "Disable"
then reboot and check again

bmpalmeida, 2021-05-24:
I'm also with asus router having the same problem.

I tryed a lot of configuration changes on the router, but none of them enable me to get the shelly devices stable.

It also happens in a shelly EM with stock firmware.

But im my case when some of the devices get unresponsive not even the physical button works (maybe this can be related or completely unrelated), and i notice they are constant rebooting (the uptime is always a couple of minutes).

And i have a lot of this message (every 2 seconds) in the log of some devices, but not on others:

607946990 mgos_rpc_channel_ua:160 Incoming frame is too big, dropping.

iliquifysnacks, 2021-05-26:
I think it could be a router issue, maybe related to something in the Asus Firmware. Giving the shelly a fixed IP in the router seems to make everything more robust.

bmpalmeida, 2021-07-12:
I've just manage to get a new log of one of the devices i have that is getting frozen (not even switch lights with the physical button works).

Maybe the device is just broken :)

15223660 shelly_debug.cpp:227    No log file, sending new entries
16393666 esp_main.c:138          SDK: pm open,type:0 0
16482605 shelly_main.cpp:543     Up 16.47, HAP 0/0/12 ns 0, RAM: 23572/38600; st 45; 0.1: st:0 in_st:0 inm:1 ininv:0
18811042 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.50.96:52361
19974133 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.50.96:52361
21201798 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.50.96:52361
21818638 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.50.96:52361
22522692 mgos_rpc_channel_ua:139 0x3fff44e4 Corrupted frame (0): '' '{&•…€ó.•' 00000000 00000000
22819433 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.50.96:52361
23941485 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.50.96:52361
24482538 shelly_main.cpp:543     Up 24.47, HAP 0/0/12 ns 0, RAM: 25468/38600; st 44; 0.1: st:0 in_st:0 inm:1 ininv:0
24813496 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.50.96:52361
25814772 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.50.96:52361
26296024 mgos_rpc_channel_ua:139 0x3fff44e4 Corrupted frame (0): '' '{H!(' 00000000 00000000
26831422 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.50.96:52361
27891400 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.50.96:52361
28930531 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.50.96:52361
29817057 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.50.96:52361
30186118 mgos_rpc_channel_ua:139 0x3fff44e4 Corrupted frame (0): '' '{† ' 00000000 00000000
32482357 shelly_main.cpp:543     Up 32.47, HAP 0/0/12 ns 0, RAM: 25204/38600; st 45; 0.1: st:0 in_st:0 inm:1 ininv:0
32569330 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.50.96:52361
33582900 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.50.96:52361
34580969 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.50.96:52361

closed 2022-12-22 by github-actions[bot]