mongoose-os-apps/shelly-homekit #673: Shelly 2.5 can not be flashed with homekit firmware

moranchoj, 2021-05-10:
Hi, I have a shelly 2.5 adquired recently. Using the shelly app i can install and works well but I am unable to flash with the homekit firmware. Fails  using the we/ota option

http://192.168.XXX.XXX/ota?url=http://rojer.me/files/shelly/shelly-homekit-Shelly25.zip

butn also fails using the flash-shelly.py command line option.

When trying to upgrade by using web/ota option, nothing happens. It remains the same stock firmware and the shelly app says that there is a new version ready to be installed and shows the homekit firmware version (see pciture attached) but it is impossible to be flashed.

When using the command line alternative, the python script says that stock firmware needs to be upgraded. You answer yes but upgrade fails. Later tries  to install the homekit firmware but also fails (see script output attached).

it seems that firmware update is blocked if it is not a stock firmware.
[image]
[image]

timoschilling, 2021-05-10:
On your screenshot I can see that your first 2.5 wasn’t able to flash from stock 0.0.0 to stock 1.10.4, and the other 2.5 wasn’t able to flash to the HomeKit firmware. For me this sounds like a network issue.

moranchoj, 2021-05-10:
If it was a network issue, shelly app should not be able to update firmware
either but the shelly app updates well the stock firmware

va escriure:

moranchoj, 2021-05-10:
Really was a network issue. I was using my router to radiate 2.4GHz wifi and my Davolo PLC to radiate only 5Ghz wifi. I created a new 2.4GHz wifi in my Davolo PLC and connected the shelly to the new 2.4Ghz wifi and everything worked well. Thanks Timo to point me to the right direction