mongoose-os-apps/shelly-homekit #674: Matter compatibility

yuvalbeck, 2021-05-12:
It would be great to prepare to make the shelly Matter `compatible :)

Alex-Bricklayer, 2022-10-19:
Is there any news about the matter support?

rionshin, 2022-10-19:
Current Matter protocol has limited list of devices, so i am not sure Matter is something we need for Shelly for now.

timoschilling, 2023-02-23:
Don't misunderstand me, I would like to have matter too. But at the moment no matter implementation running on a esp8266 the cpu of the not plus / pro devices. That means matter isn't possible at the moment. Maybe it's change in the future then we can rethink.

For the plus / pro devices Shelly announced matter support and bringing matter to our firmware is a huge amount of work, so there is no relation between benefit and cost of dev time.