mongoose-os-apps/shelly-homekit #677: Reset a shelly 1 in the System menu

Kelmis13, 2021-05-14:
I accidentally reset a shelly 1 in the System menu. Now the shelly is no longer accessible via the IP.

timoschilling, 2021-05-14:
The reset Button in the system section resets the WiFi credentials. Take a look for a WiFi named shelly1-xxxx and open 192.168.33.1 and resetup the WiFi credentials

Kelmis13, 2021-05-15:
> The reset Button in the system section resets the WiFi credentials. Take a look for a WiFi named shelly1-xxxx and open 192.168.33.1 and resetup the WiFi credentials

That worked fine - thank you very much!
After entering the wrong Wifi name (upper case / lower case), I had to manually reset the Shelly again. Power off/on and 5 x push the switch.

closed 2021-05-15 by timoschilling