mongoose-os-apps/shelly-homekit #679: Add Video to Doorbell added in 2.8.

victor-vz, 2021-05-16:
I think this one might be tricky. Will it be possible to integrate a Video stream to the Doorbell to be able to receive rich notifications with Video stream? As you already made the HomePod ring somehow integrating for e.g. a Eufy HKSV camera into the shelly doorbell might even be able to announce the person at the door. I saw a similar request for homebridge: [https://www.reddit.com/r/homebridge/comments/jl320a/wanting_eufy_indoor_cam_2k_to_trigger_a_doorbell/](url)
[https://github.com/Sunoo/homebridge-camera-ffmpeg#optional-parameters](url)
[image]
[image]
Love the work u guys are doing!!!

victor-vz, 2021-05-16:
Also not an issue but would be a great feature!

jsiegenthaler, 2021-05-17:
A Homebridge plugin exists that does just this: https://github.com/Sunoo/homebridge-camera-ffmpeg

It needs the processing power of a raspberry pi, and I cannot for the life of me see that ever happening on a Shelly!

I am guessing this is orders of magnitude too complex a task for a Shelly to perform.

victor-vz, 2021-05-17:
Mhm, that is unfortunate. In iOS 12 if a motion sensor was triggered and a camera was in the same room the notification from the motion sensor would also include a screenshot from the camera. This changed in iOS 13 and now the camera and motion sensor need to be the same device (not only in the same room) to receive the notification with the screenshot. Would it be possible to group a HKSV camera with the shelly doorbell/motion sensor. Logitechs Circle View Camera has an PIR sensor integrated which you can Group and Ungroup. What would you need to do/change to make HomeKit think that the Camera and the Shelly are the same device?

rojer, 2021-05-17:
right, video is completely out of the question for esp8266, i'm afraid

jsiegenthaler, 2021-05-17:
@victor-vz I use a Shelly1 and the homebridge video ffmpeg plugin to do exactly what you want: have a video doorbell.
but you need to do it on a raspberry pi (or some other always-on server) running homebridge

victor-vz, 2021-05-17:
@jsiegenthaler Are you still able to use the Homekit camera (Eufy 2K Indoor) the same way with HomeKit Secure Video and have the additional Videodoorbell? Does the Videodoorbell also leverage the facial recognition for the announcements via HomeKit? Is there a step for step guide you can recommend?

jsiegenthaler, 2021-05-17:
@victor-vz I didn't use the Eufy video stream, I have a different (static) video stream as the source. But you can use any rtsp video stream with the ffmpeg plugin. 
Because my video stream is static (a GIF file), my HomeKit doesn't get any faces it can recognise.

And I do not know about HKSV, I do not use HKSV. I suggest you try it and see.

closed 2021-05-17 by victor-vz