mongoose-os-apps/shelly-homekit #682: Unable to access web interface after "revert to stock" with password set

skeltavik, 2021-05-17:
Hi all,

So my Shelly 1 had the new firmware that allows to set a password to prevent unauthorized access (v2.9.0). As far as I know it works with digest auth from what I saw in the code and HTTP request. 

But today I tried to revert it back to the stock firmware by clicking "Revert to stock". It uploaded and I tried to reconnect to the same IP and was greeted with a typical username/password dialog in my browser.

[image]

I tried "admin" and my set-up password, but without any luck. I also tried the 5-times switch reset, that didn't work either. Any idea what I can do in order to have access again? 

Thanks in advance!

skeltavik, 2021-05-17:
I also tried using cURL, but that didn't do much either:

`curl --digest -u admin:examplepassword -v "http://192.168.1.168/"`

```
*   Trying 192.168.1.168...
* TCP_NODELAY set
* Connected to 192.168.1.168 (192.168.1.168) port 80 (#0)
* Server auth using Digest with user 'admin'
> GET / HTTP/1.1
> Host: 192.168.1.168
> User-Agent: curl/7.64.1
> Accept: */*
>
< HTTP/1.1 401 Unauthorized
< WWW-Authenticate: Digest qop="auth", realm="shelly1-ED6E8C", nonce="60<stripped>", algorithm=MD5
< Content-Length: 0
<
* Connection #0 to host 192.168.1.168 left intact
* Issue another request to this URL: 'http://192.168.1.168/'
* Found bundle for host 192.168.1.168: 0x7fa1936050f0 [can pipeline]
* Could pipeline, but not asked to!
* Re-using existing connection! (#0) with host 192.168.1.168
* Connected to 192.168.1.168 (192.168.1.168) port 80 (#0)
* Server auth using Digest with user 'admin'
> GET / HTTP/1.1
> Host: 192.168.1.168
> Authorization: Digest username="admin", realm="shelly1-ED6E8C", nonce="60<stripped>", uri="/", cnonce="YTVjNGM<stripped>", nc=00000001, qop=auth, response="790f10d<stripped>", algorithm="MD5"
> User-Agent: curl/7.64.1
> Accept: */*
>
< HTTP/1.1 401 Unauthorized
* Authentication problem. Ignoring this.
< WWW-Authenticate: Digest qop="auth", realm="shelly1-ED6E8C", nonce="60<stripped>", algorithm=MD5
< Content-Length: 0
<
* Connection #0 to host 192.168.1.168 left intact
* Closing connection 0
```

timoschilling, 2021-05-17:
Idea for a short fix flash it with the HomeKit firmware, set the password to “” to disable it and then revert to stock again.

skeltavik, 2021-05-17:
Hi @timoschilling, I can't flash the firmware anymore either via HTTP. 

```
╰─$ curl -v "http://192.168.1.168/ota?url=http://rojer.me/files/shelly/shelly-homekit-Shelly1.zip"
*   Trying 192.168.1.168...
* TCP_NODELAY set
* Connected to 192.168.1.168 (192.168.1.168) port 80 (#0)
> GET /ota?url=http://rojer.me/files/shelly/shelly-homekit-Shelly1.zip HTTP/1.1
> Host: 192.168.1.168
> User-Agent: curl/7.64.1
> Accept: */*
>
< HTTP/1.1 401 Unauthorized
< WWW-Authenticate: Digest qop="auth", realm="shelly1-ED6E8C", nonce="60a2b816", algorithm=MD5
< Content-Length: 0
<
* Connection #0 to host 192.168.1.168 left intact
* Closing connection 0
```

It returns a "401 Unauthorized" unfortunately.

timoschilling, 2021-05-17:
you could try this:
1. Disconnect the power from the Shelly by  turning off a breaker or other means. 
2. Restore power.
3. Within the  first minute, toggle a switch connected to the "SW" input on the Shelly 5  times on/off (10 total) or press and release momentary button 5 times.

timoschilling, 2021-05-17:
@rojer any ideas how we can fix this?

rojer, 2021-05-17:
ugh, nasty. i'm taking a look.

rojer, 2021-05-17:
ok, i see what the problem is. i will focus on fixing it in the code to prevent from happening for new users but i'm afraid device affected by this may not be recoverable other than by serial connection...

rojer, 2021-05-18:
ok, fix has been committed and i'm pushing 2.9.1 now.
@skeltavik unfortunately, i don't think it's possible to recover the device without a serial cable.
setup is described [here](https://github.com/mongoose-os-apps/shelly-homekit/wiki/Serial-connection).
i can offer to unbrick the device (or devices) for you if you send it (them) to me, please contact me at rojer@rojer.me if you'd like me to do that.
please accept my apologies for the inconvenience caused.

rojer, 2021-05-18:
ok, 2.9.1 is out

closed 2021-05-18 by rojer