mongoose-os-apps/shelly-homekit #683: 2.5PM switch status via REST API / GET?

thncode, 2021-05-17:
How can I request this / is this possible?

cheers,
 Thomas

timoschilling, 2021-05-17:
It’s possible /rpc/Shelly.GetInfoExt there you find `components[0][“state”]`

thncode, 2021-05-17:
thx a lot - where is this documented???

timoschilling, 2021-05-17:
Looks like it’s not documented at the moment. The rest of the api docs are here https://github.com/mongoose-os-apps/shelly-homekit/wiki/Development#API

@andyblac would you like to improve the wiki there?

thncode, 2021-05-17:
sorry, one more question: is relay state toggle possible as well?

andyblac, 2021-05-17:
> sorry, one more question: is relay state toggle possible as well?

unfortuantly no, the api does not have a toggle action.

andyblac, 2021-05-17:
> thx a lot - where is this documented???

try something
```
andrews-mbp-wifi:tools andrewblackburn$ curl -s http://andys-bedroom-sw.local/rpc/Shelly.GetInfoExt | jq  --raw-output '.components[0].state'
false
```

timoschilling, 2021-05-19:
I think we can close this for now, question is answered and google feature request has his own issue somewhere

closed 2021-05-19 by timoschilling