mongoose-os-apps/shelly-homekit #684: Doorbell not supported by HomeKit

Dabr422, 2021-05-18:
Hi, I’ve tried to configure my latest Shelly for doorbell mode and get it to work. Sort of…
The unit shows as two parts, a switch and a doorbell. If I close the input circuit I get a notification as intended. But the doorbell part shows in HomeKit as “not supported”. What am I doing wrong. I attach a screen dump of my HomeKit page and the config of the Shelly. I’m running v2.9.0. Thanks

Daniel

[image]
[image]
[image]

jsiegenthaler, 2021-05-18:
That's the way it is. A simple Doorbell will provide notifications, but is not supported (any more). Only video doorbells are supported. If you have a raspberry pi and Homebridge, use camera-ffmpeg to make a video doorbell using a gif or still image as the video part. That's what I do.

Dabr422, 2021-05-18:
Thanks for for quick reply. I suspected it was like that but it’s good to get it confirmed.

closed 2021-05-19 by timoschilling