mongoose-os-apps/shelly-homekit #686: Fashscript 3.0.0 - beta testers wanted.

andyblac, 2021-05-20:
hi guys,

i am after some people to help beta test my script version 3.0.0, it had a major rewrite, and now supports saving of your own defaults and profiles.

to save new defaults just do your commandline options you want to be your new defaults and add `--save-defaults` to end, to use them simply run script as normal `./flash-shelly.py`
i.e.
`./flash-shelly.py -l -m keep --save-defaults` this will make the script run in list only and check for updates for both homekit and stock.

to use profiles use `--save-config SAVE_CONFIG`  SAVE_CONFIG being profile name.
i.e.
`./flash-shelly.py -f -m keep -ft stock -y --save-config stock` and run it by `./flash-shelly.py --config stock`, this example will only check stock devices for stock firmware updates and any are found update them automatically.

you can have as many profiles as you want, you can even save manual IP addresses and host names to a profile, so if the scanner does not work for you, (ie your shelly's are on a differan vlan) you can now save them as a profile to load them.

i.e.
`./flash-shelly.py 192.168.3.40 Office-Window-Strip.local --save-config manualhosts`, and to run it `./flash-shelly.py -config manualhosts`

.you can also add any additional commandline options to profiles to override saved options.
doing `./flash-shelly.py -config manualhosts -l` would just updates for example.

please use `./flash-shelly.py --help` to see all availble options.

i would really like your feedback on this, and thanks for help in testing
[flash-shelly.-3.0.0-beta1.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/6515415/flash-shelly.-3.0.0-beta1.zip)

kachyy, 2021-05-22:
used only for update all (python3 flash-shelly.py -am keep)
worked great on all devices

andyblac, 2021-05-28:
update with beta 4,
in beta 4 you can now flash a particular version direct from stock to HomeKit with `-V x.x.x` x.x.x being the version you want to flash.

trendchiller, 2021-08-04:
I'm using Ubuntu 18.04.
I installed zeroconf with apt install python3-zeroconf but get the following error - does anyone have some hint ?

Scanning for Shelly devices...
Exception in thread zeroconf-ServiceBrowser__http._tcp.local.:
Traceback (most recent call last):
  File "/usr/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3/dist-packages/zeroconf.py", line 1334, in run
    handler(self.zc)
  File "/usr/lib/python3/dist-packages/zeroconf.py", line 1282, in <lambda>
    state_change=state_change,
  File "/usr/lib/python3/dist-packages/zeroconf.py", line 1197, in fire
    h(**kwargs)
  File "/usr/lib/python3/dist-packages/zeroconf.py", line 1255, in on_change
    listener.add_service(*args)
  File "flash-shelly.py", line 224, in add_service
    wifi_ip = socket.inet_ntoa(info.addresses[0])
AttributeError: 'ServiceInfo' object has no attribute 'addresses'

andyblac, 2021-08-04:
python 3.6 is old, please try update to 3.9 and update the zeroconf module as well.

andyblac, 2021-08-04:
zeroconf should auto install, but the correct way to manually install is

`pip3 install zeroconf`

trendchiller, 2021-08-06:
Hi !
You were right !
The python version did the magic.
Using python 3.7.3 made it work.
Thank you very much !

juangamnik, 2021-11-13:
Dear @andyblac I would like to be beta tester, if the flash-script (never used it) is able to trigger an update on device. My use case would be to have a cron job which regularly checks whether an update is available for an device, tries the update (with rollback option on), tries to reach the web view (or even better reach for a self-Health-Check) and if this works it commits the update. If an update did not work it rollbacks automatically after some time (as when I deploy the update during development) and sends me a notification (so of course this is not part of the flash-script functionality, but of my script.

andyblac, 2021-11-13:
here is the latest beta of version, for an automatic update of all devices (stock or homekit) you could use this command line in your cronjob
`./flash-shelly.py -faym keep`

that will check all devices on your network and see if any updates are available, and if so update them automatically.

if you want just to see if they are any updates but not apply them run

`./flash-shelly.py -am keep`

use `./flash-shelly.py -h` to see a list of all options that the script can do.
[flash-shelly-3.0.0-beta6.zip](https://github.com/mongoose-os-apps/shelly-homekit/files/7532155/flash-shelly-3.0.0-beta6.zip)

bmpalmeida, 2021-12-17:
I have 40 devices in my network, and every time i run the script to update them all its very rare that all of them gets updated at first try, some timeout, some are not discovered.

normally i run the script in a while loop, to ensure that everything gets updated.

Could it be interesting to have a --loop flag to have it running forever, or something like an --ensure-update that cycles until everything is with the last version?

andyblac, 2021-12-17:
not something I plan on doing, it sounds like network issues (maybe mesh devices). anyway, tbh, i'am about done with the script (ver 3.0), it more than suits my needs, but by all means, take a fork and add stuff as you see fit. anyway let get back on topic here.

bmpalmeida, 2021-12-17:
Indeed mesh network is always a nightmare, never been 100% reliable.

I will try to get some time and code something.

closed 2022-01-07 by rojer