mongoose-os-apps/shelly-homekit #687: Cannot Flash Shelly 2.5

yuvalbeck, 2021-05-23:
Hi I cannot flash new shelly 2.5 with stock firmware - maybe new version from Shelly.
[debug.log](https://github.com/mongoose-os-apps/shelly-homekit/files/6527930/debug.log)

Thank you

andyblac, 2021-05-23:
hi,

for some strange reason the device is not seeing the update on stock servers, is the device using DHCP or Static IP, please double check it has an working internet conection.

```
2021-05-23 13:30:17,439 DEBUG 622 Remote Download URL: http://shelly-api-eu.shelly.cloud/firmware/SHSW-25.zip
2021-05-23 13:30:17,439 DEBUG 627 requests.get(url=http://10.0.1.165/ota?update=true, auth=('None', 'None')
2021-05-23 13:30:17,470 TRACE 629 {"status":"unknown","has_update":false,"new_version":"","old_version":"20191217-074901/1047-long-id-for-shelly-devices@ab2bdce7"}
```

yuvalbeck, 2021-05-23:
> hi,
>
> for some strange reason the device is not seeing the update on stock servers, is the device using DHCP or Static IP, please double check it has an working internet conection.
>
> ```
> 2021-05-23 [...]

I think you are right in the stock firmware web UI there is no time and not able to connect the shelly's cloud, so I believe no internet connection (other devices on network has internet connection) in the shelly setting I didn't set an static IP but my router automatically assigns static IP for every device in the network

yuvalbeck, 2021-06-06:
So I have a very strange problem maybe someone can help.

I replaced one of my shelly 2.5 devices that was connect to 2 lights because I couldn't control one of the lights, even the physical switch stopped working.

In the beginning the new shelly fixed the issue but It won't connect to the internet although it is connected to my home network. after a week I am have huge delay in the response time when I try to turn on/off the lights from the physical buttons and sometime no response at all.

I also ping the device directly from the router console and I have 92% packet loss.

Does anyone have an idea what causes those issues ?

Thank you!

closed 2021-07-15 by github-actions[bot]

ebabel, 2022-06-19:
I read that you can download and host the file locally, then force update pointing to locally hosted file, have you tried that? I'm also getting this error and might try it soon.

joschkarick, 2023-01-16:
Thanks, the local file server did the trick for me. I used "python3 -m http.server" to create the file server. [Credits](https://sequr.be/blog/2020/11/force-update/downgrade-shelly-firmware/)

andyblac, 2023-01-17:
@joschkarick the script can flash local files to the shelly device just so you know.