mongoose-os-apps/shelly-homekit #688: Shelly Dimmer 2 Support

guymorgen88, 2021-05-24:
Hi all,

Is Shelly Dimmer 2 support?
I saw the table  "Summary of supported devices and features" but I'm not sure if it's 100% ok.

Thanks for your help

timoschilling, 2021-05-24:
The Dimmer2 is not supported at the moment, feature request is in #14. 

Simulator device, depends on your needs. If you need to dimm 110-240V there is no alternative. If you can dimm 12-24V you can use the RGBW2 which is supported. Or you use one of the Shelly bulbs which will be supported in the next 1-2 month.

closed 2021-05-24 by timoschilling

guymorgen88, 2021-05-24:
Hey Tim, 

I need a dime for 110-240V. Any estimation when will the Dimmer 2 be supported?

timoschilling, 2021-05-24:
No

sg-generix, 2021-11-05:
> The Dimmer2 is not supported at the moment, feature request is in #14.
>
> Simulator device, depends on your needs. If you need to dimm 110-240V there is no alternative. If you can dimm 12-24V you can [...]

Hello, thank you for great support @timoschilling 
Do you have idea when will be ready the support of shelly bulbs?