mongoose-os-apps/shelly-homekit #695: Lock with functional input

stoex, 2021-06-02:
Hey there,

I am using my shelly 1 to unlock my front door. Is it possible to detect wether the door is opened or closed?

I thought about the input of the shelly (just like the function as a garage door opener). It would be very nice if you just can put a magnetic sensor to the door.

The question is if you need a second icon in the homekit app?

Thanks.

timoschilling, 2021-06-02:
It’s already possible, use like you did it right now, but set the input to detached, the reload the UI and now you can use the input with a magnetic sensor as contact sensor.

timoschilling, 2021-06-05:
I think the question is answered, if you have any additional question feel free to ask to reopen the issue.

closed 2021-06-05 by timoschilling