mongoose-os-apps/shelly-homekit #696: Shelly 1L stuck in Loading screen

Jasko65, 2021-06-02:
Hello, 

it a mistake! Instate of entering the WIFI PW in right section I put in the security one. So my shelly is stuck in UI with the loading screen. I tried to reset it with "--local -file SHSW-L.zip" but it always says "unrecognized arguments: --local -file". So is there any chance to reset it without internet or to get it out loading screen?

BG

andyblac, 2021-06-02:
> Hello,
>
> it a mistake! Instate of entering the WIFI PW in right section I put in the security one. So my shelly is stuck in UI with the loading screen. I tried to reset it with "--local -file [...]

to use local file it's `--local-file xxxx.zip`

Jasko65, 2021-06-02:
> > Hello,
> > it a mistake! Instate of entering the WIFI PW in right section I put in the security one. So my shelly is stuck in UI with the loading screen. I tried to reset it with "--local -file SHSW-L.zip" but it always says "unrecognized arguments: --local -file". So is there any chance to reset it without internet or to get it out loading screen?
> > BG
> 
> to use local file it's `--local-file xxxx.zip` (no space between --local and -file), but it will still require access to the shelly via the IP address, the only way to refresh without using an IP is to flash via serial connection.

I have access via its own wifi but without internet connection. I can communicate with 192.168.33.1 but via Terminal it won't find it.

andyblac, 2021-06-02:
can you login UI when on own wifi ?

Jasko65, 2021-06-02:
No it shows me it is loading. And i can't do anything. I also tried to restart it but without success.

andyblac, 2021-06-02:
ok, it seems the only way is to refresh by serial

andyblac, 2021-06-02:
do you not get the password prompt ?

Jasko65, 2021-06-02:
Okay. Thank you for your support!

closed 2021-06-02 by Jasko65

rojer, 2021-06-02:
> I can communicate with 192.168.33.1

when you open http://192.168.33.1/ does the UI load?

Jasko65, 2021-06-02:
> do you not get the password prompt ?

no
> > I can communicate with 192.168.33.1
> 
> when you open http://192.168.33.1/ does the UI load?

no it just tells me it is loading

Jasko65, 2021-06-02:
I just found a solution for me. I held the reset button. Now it seems working

rojer, 2021-06-02:
great, thanks

rojer, 2021-06-02:
if you ever need to recover a device, it's documented here - https://github.com/mongoose-os-apps/shelly-homekit/wiki/Recovery