mongoose-os-apps/shelly-homekit #698: Shelly 2 - Pulsating output on AC input operation

omerduzyol, 2021-06-07:
Hi,

First of all, thanks to everyone who contributed to this lovely project! I made a setup to control the speed of MVHR ventilation unit using my Shelly 2 controller. Please refer to diagram below:

[image]

Used the recent version of homekit firmware and configured both switches like below:

```
HAP Service Type: Switch
Input Mode: Toggle
Initial State: Input
Auto Off: False
```

Everything works well if override switches are not connected or, turned on (closed) but when I turn off the override switches (open circuit) output starts pulsating and I can hear that from the relay contacts very quickly toggling. I'm not sure if I'm doing something wrong here or is that related to Shelly 2 etc.

Any help appreciated.

Thanks!

timoschilling, 2021-06-07:
Only to be sure, do you really have Shelly2 or a Shelly2.5?

timoschilling, 2021-06-07:
Your diagram LGTM, but `when I turn off the override switches (open circuit) output starts pulsating` sound like there is a mistake in your wiring or the diagram didn't match the reality.

omerduzyol, 2021-06-07:
> Only to be sure, do you really have Shelly2 or a Shelly2.5?

it's shelly 2, not 2.5 :) 

I checked all the wiring and it seemed there is no other connections other than the override switches.

rojer, 2021-06-07:
can you open the debug log (link in the "System" section of the UI) and reproduce the issue? then post the log here, i'd like to take a look.

omerduzyol, 2021-06-08:
@rojer please find the log below:
```
6020242123 shelly_debug.cpp:227    No log file, sending new entries
6027739154 shelly_main.cpp:543     Up 6027.72, HAP 0/3/12 ns 3, RAM: 21408/38664; st 0; 0.1: st:1 in_st:0 inm:3 ininv:0; 3.1: svt:3 inv:0 st:0 m:0 lea: -1.000; 0.2: st:0 in_st:1 inm:3 ininv:0; 10.2: svt:10 inv:0 st:1 m:0 lea: -1.000
6034129091 shelly_input.cpp:76     Input 1: change (state 1)
6034161127 shelly_input.cpp:76     Input 1: change (state 0)
6034208469 shelly_input.cpp:76     Input 1: change (state 1)
6034241280 shelly_input.cpp:76     Input 1: change (state 0)
6034246570 shelly_input.cpp:76     Input 1: double (state 0)
6034251673 shelly_hap_stateles:166 Input 1: HAP event (mode 0): 1
6034301715 shelly_input.cpp:76     Input 1: change (state 1)
6034360996 shelly_input.cpp:76     Input 1: change (state 0)
6034408470 shelly_input.cpp:76     Input 1: change (state 1)
6034441374 shelly_input.cpp:76     Input 1: change (state 0)
6034446580 shelly_input.cpp:76     Input 1: double (state 0)
6034451679 shelly_hap_stateles:166 Input 1: HAP event (mode 0): 1
6034501869 shelly_input.cpp:76     Input 1: change (state 1)
6034560973 shelly_input.cpp:76     Input 1: change (state 0)
6034608149 shelly_input.cpp:76     Input 1: change (state 1)
6034641431 shelly_input.cpp:76     Input 1: change (state 0)
6034646653 shelly_input.cpp:76     Input 1: double (state 0)
6034651747 shelly_hap_stateles:166 Input 1: HAP event (mode 0): 1
6034701962 shelly_input.cpp:76     Input 1: change (state 1)
6034761148 shelly_input.cpp:76     Input 1: change (state 0)
6034808801 shelly_input.cpp:76     Input 1: change (state 1)
6034841547 shelly_input.cpp:76     Input 1: change (state 0)
6034846798 shelly_input.cpp:76     Input 1: double (state 0)
6034851889 shelly_hap_stateles:166 Input 1: HAP event (mode 0): 1
6034901914 shelly_input.cpp:76     Input 1: change (state 1)
6034981094 shelly_input.cpp:76     Input 1: change (state 0)
6035010093 shelly_input.cpp:76     Input 1: change (state 1)
6035063194 shelly_input.cpp:76     Input 1: change (state 0)
6035068406 shelly_input.cpp:76     Input 1: double (state 0)
6035073500 shelly_hap_stateles:166 Input 1: HAP event (mode 0): 1
6035102528 shelly_input.cpp:76     Input 1: change (state 1)
6035161470 shelly_input.cpp:76     Input 1: change (state 0)
6035208844 shelly_input.cpp:76     Input 1: change (state 1)
6035241605 shelly_input.cpp:76     Input 1: change (state 0)
6035246850 shelly_input.cpp:76     Input 1: double (state 0)
6035251941 shelly_hap_stateles:166 Input 1: HAP event (mode 0): 1
6035302589 shelly_input.cpp:76     Input 1: change (state 1)
6035403440 shelly_input.cpp:76     Input 1: change (state 0)
6035462055 shelly_input.cpp:76     Input 1: change (state 1)
6035521136 shelly_input.cpp:76     Input 1: change (state 0)
6035526294 shelly_input.cpp:76     Input 1: double (state 0)
6035531381 shelly_hap_stateles:166 Input 1: HAP event (mode 0): 1
6035562622 shelly_input.cpp:76     Input 1: change (state 1)
6035621055 shelly_input.cpp:76     Input 1: change (state 0)
6035669904 shelly_input.cpp:76     Input 1: change (state 1)
6035701289 shelly_input.cpp:76     Input 1: change (state 0)
6035706581 shelly_input.cpp:76     Input 1: double (state 0)
6035711671 shelly_hap_stateles:166 Input 1: HAP event (mode 0): 1
6035739024 shelly_main.cpp:543     Up 6035.72, HAP 0/3/12 ns 3, RAM: 19812/38664; st 0; 0.1: st:1 in_st:0 inm:3 ininv:0; 3.1: svt:3 inv:0 st:0 m:0 lea: 0.010; 0.2: st:0 in_st:1 inm:3 ininv:0; 10.2: svt:10 inv:0 st:1 m:0 lea: -1.000
6035769413 shelly_input.cpp:76     Input 1: change (state 1)
6035801184 shelly_input.cpp:76     Input 1: change (state 0)
6035848244 shelly_input.cpp:76     Input 1: change (state 1)
6035881718 shelly_input.cpp:76     Input 1: change (state 0)
6035886981 shelly_input.cpp:76     Input 1: double (state 0)
6035892062 shelly_hap_stateles:166 Input 1: HAP event (mode 0): 1
6035942223 shelly_input.cpp:76     Input 1: change (state 1)
6036001402 shelly_input.cpp:76     Input 1: change (state 0)
6036048484 shelly_input.cpp:76     Input 1: change (state 1)
6036081620 shelly_input.cpp:76     Input 1: change (state 0)
6036086859 shelly_input.cpp:76     Input 1: double (state 0)
6036091957 shelly_hap_stateles:166 Input 1: HAP event (mode 0): 1
6036142348 shelly_input.cpp:76     Input 1: change (state 1)
6036201212 shelly_input.cpp:76     Input 1: change (state 0)
6036248709 shelly_input.cpp:76     Input 1: change (state 1)
6036281631 shelly_input.cpp:76     Input 1: change (state 0)
6036286858 shelly_input.cpp:76     Input 1: double (state 0)
[... 92 lines cut ...]
6039481331 shelly_input.cpp:76     Input 1: change (state 0)
6039528975 shelly_input.cpp:76     Input 1: change (state 1)
6039561859 shelly_input.cpp:76     Input 1: change (state 0)
6039567044 shelly_input.cpp:76     Input 1: double (state 0)
6039572139 shelly_hap_stateles:166 Input 1: HAP event (mode 0): 1
6039768983 shelly_input.cpp:76     Input 1: change (state 1)
6039801522 shelly_input.cpp:76     Input 1: change (state 0)
6039849019 shelly_input.cpp:76     Input 1: change (state 1)
6039881457 shelly_input.cpp:76     Input 1: change (state 0)
6039886701 shelly_input.cpp:76     Input 1: double (state 0)
6039891788 shelly_hap_stateles:166 Input 1: HAP event (mode 0): 1
6040069022 shelly_input.cpp:76     Input 1: change (state 1)
6040101574 shelly_input.cpp:76     Input 1: change (state 0)
6040149115 shelly_input.cpp:76     Input 1: change (state 1)
[... cut ...]
6059739225 shelly_main.cpp:543     Up 6059.72, HAP 0/3/12 ns 3, RAM: 21408/38664; st 0; 0.1: st:1 in_st:0 inm:3 ininv:0; 3.1: svt:3 inv:0 st:0 m:0 lea: 19.529; 0.2: st:0 in_st:1 inm:3 ininv:0; 10.2: svt:10 inv:0 st:1 m:0 lea: -1.000
6067739127 shelly_main.cpp:543     Up 6067.72, HAP 0/3/12 ns 3, RAM: 21408/38664; st 0; 0.1: st:1 in_st:0 inm:3 ininv:0; 3.1: svt:3 inv:0 st:0 m:0 lea: 27.529; 0.2: st:0 in_st:1 inm:3 ininv:0; 10.2: svt:10 inv:0 st:1 m:0 lea: -1.000
```
I detached the input and configured it like below, otherwise, it resets the config

[image]

timoschilling, 2021-06-08:
Looks like an hardware issue or a miss configured input. Are you sure that you have an momentary switch? Have you tried the other switch modes?

omerduzyol, 2021-06-09:
> Looks like an hardware issue or a miss configured input. Are you sure that you have an momentary switch? Have you tried the other switch modes?

Actually, I was messing around with different switch modes to see if there will be any difference or not. 

> looks like some major noise on the input. doesn't happen with input 2, right?

That's what I was thinking as well. The wire is not connected anywhere when all the override switches are off (open circuit). Would that be an EMF messing with the wiring? I suppose that should be filtered when Shelly operating in 220 AC mode.

omerduzyol, 2021-06-09:
Forgot to mention, I don't have input 2 connected to any wire and disabled. But that doesn't happen even if I activate it. Didn't try to switch inputs tbh. I also ordered a new Shelly 2.5 and will give a try with that see how that works.

closed 2021-07-17 by github-actions[bot]