mongoose-os-apps/shelly-homekit #700: Shelly 2.5 in Garage Door Mode doesn't work, restarts every time

menzo, 2021-06-08:
I have 2 garage doors, one with a Shelly1 and one with a Shelly 2.5, both with Shelly-homekit 2.9.1 FW.

While the Shelly1 works perfectly, the 2.5 works only first time after a power-reset, after that it doesn't switch, but resets every time when I try to operate the door. 

- They have the same, common 24V power source from one of the motors
- When I open/close the Shelly1, the 2.5 doesn't reset
- When I try to open/close the Shelly 2.5, the Shelly1 doesn't reset
- When I operate the 2.5's door with the remote, the 2.5 doesn't reset
- The Shellys are next to each other, has the same RSSI -74 (from Ubiquity AC Pro wifi)
- When I completely disconnect the Shelly1 and the other door, the problem is still the same with the 2.5
- When I don't try to open/close with the 2.5 it doesn't resets, 
- When I do a soft reset on the 2.5 (from the web interface) nothing changes, still not working
- When I disconnect and reconnect the power from the 2.5 it works perfectly, no restarts, but only one time

I think its a very similar issue like this one:
https://github.com/mongoose-os-apps/shelly-homekit/issues/251

timoschilling, 2021-06-08:
Can you post some logs from during the restart please

menzo, 2021-06-08:
Sure:

> 5755765013 shelly_debug.cpp:227    No log file, sending new entries
5763487271 shelly_main.cpp:543     Up 5763.48, HAP 0/6/12 ns 6, RAM: 20696/38440; st 65; 5.1: cur:open tgt:open cl:0 op:-1
5771487316 shelly_main.cpp:543     Up 5771.48, HAP 0/6/12 ns 6, RAM: 20696/38440; st 65; 5.1: cur:open tgt:open cl:0 op:-1
5779487120 shelly_main.cpp:543     Up 5779.48, HAP 0/6/12 ns 6, RAM: 20696/38440; st 65; 5.1: cur:open tgt:open cl:0 op:-1
5787486861 shelly_main.cpp:543     Up 5787.48, HAP 0/6/12 ns 6, RAM: 20696/38440; st 65; 5.1: cur:open tgt:open cl:0 op:-1
5795486874 shelly_main.cpp:543     Up 5795.48, HAP 0/6/12 ns 6, RAM: 20696/38440; st 65; 5.1: cur:open tgt:open cl:0 op:-1
5796150284 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.0.38:62661
5797147887 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.0.38:62661
5798152486 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.0.38:62661
5799150840 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.0.38:62661
5800143171 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 192.168.0.38:62661
5800647814 mg_rpc.c:305            Shelly.SetState via WS_in 192.168.0.38:62661
5800654117 shelly_output.cpp:63    Output 1: off -> on (GDO:close)
5800661035 shelly_hap_garage_d:307 GDO 1: Tgt State: open -> closed (0 -> 1) (RPC)
5800667682 shelly_hap_garage_d:222 GDO 1: Cur State: open -> closing (0 -> 3)

...and it restarts, but nothing else happens.

192.168.0.38 is my AppleTV 4k

timoschilling, 2021-06-08:
I don't believe that 192.168.0.38 is your Apple TV, I think it's your PC/Mac/iPhone `Shelly.GetInfoExt` is only called from the web UI.

timoschilling, 2021-06-11:
Just to be sure, does you mean restart or reset? Sometimes you wrote restart, sometimes reset.

menzo, 2021-06-11:
Restart.

Pixel-Chris, 2021-06-24:
Same issue here... but the reason:  at 24V the shelly 2.5 is unstable 

But with firmware 2.73 it is more stable

tohmc, 2021-07-06:
Same problem here.
Any news to this?

timoschilling, 2021-07-08:
@rojer any idea’s?

rojer, 2021-07-09:
but some kind of supply instability is definitely my suspicion, since it conicides with start of movement, this is when power spikes happen.
having serial logs would help immensely.
after restart, is there a core dump generated? check `http://a.b.c.d/debug/core`, is there anything there?

marcus-aa, 2021-07-27:
Shelly 2.5 is not stable with nominal voltage 24VDC.  If you add an external power supply adjusted to 27-29VDC all the problems are gone.

janernsting, 2021-08-18:
My Shelly 2.5 also reset/rebooted itself when operating on 24V DC supplied by the garage door opener. I added an external power supply (30V DC) and isolated the Shelly 2.5 with additional relays switches.

timoschilling, 2021-08-18:
I will close this, because everything looks like a hardware issue and the request core dumb hasn’t been provided

timoschilling, 2021-08-18:
But feel free to post updates if there any new hints for a software problem

closed 2021-08-18 by timoschilling

Bobbyhax, 2021-09-03:
Just want to add a note from my experience with the Shelly 2.5 and HomeKit firmware

With the shelly-homekit firmware, i minimum need 26VDC to have stable operation (on 24,5V it's unstable, anything under that, it will go into a reboot loop.
With stock firmware, I can use the power supply from my garage door opener that is just 22.5VDC

JPM87, 2022-11-24:
> My Shelly 2.5 also reset/rebooted itself when operating on 24V DC supplied by the garage door opener. I added an external power supply (30V DC) and isolated the Shelly 2.5 with additional relays switches. Since then it worked like a charm!

Hi can you provide a link to the additional relay you used? I want to use 30V too

tohmc, 2022-11-25:
I used 2 times 
"Finder 49.61.9.024.0050" 
And
1 time "MeanWell HDR-15-24"
Worked Perfect!

JPM87, 2022-11-30:
Thanks, I thought 24v ist not enough to power the shelly. Ordered an external power supply 24v and a keenso 2 Channel relais module 24v and now the shelly is stable.