mongoose-os-apps/shelly-homekit #701: I need to reset but long press does not seem to work

bybymax, 2021-06-09:
Hi guys ! 

I’ve used the shelly 2.5 with the homekit firmware, and it worked perfectly fine. Problem is : I had to change my internet provider and I thought I could use the same name and password for my wifi network, but my new router does not allow certain signs… Thus, I cannot reach my shelly anymore.
How can I reset and add it again to my shelly app before moving it back to homekit ? As I have not been able to revert the firmware before the switch between providers.
I ve read the forum but did not find a post (I probably missed it) where I could find a solution to my problem.
If you have a way for me to use my shelly again ;) I’d be super grateful !

Thanks in advance !

Bye !

Cyrille.

timoschilling, 2021-06-09:
please read https://github.com/mongoose-os-apps/shelly-homekit/wiki/Recovery

bybymax, 2021-06-12:
Hi guys !

Thank you so much, it worked like a charm. As it was the first time I would do that, I needed a little help and I’m grateful for that. Next time, I’ll be able to do it all by myself !

You guys rock !

Enjoy the week-end !

Amicalement,

Cyrille BERTRAND

closed 2021-06-12 by timoschilling