mongoose-os-apps/shelly-homekit #703: HomeKit Gate opener doesn't show correct status

TonyJacobi, 2021-06-11:
Picture 1 and 2 is of our Motorline MC2 gate opener. I have connected a Shelly 1 PM with latest 2.91 Homekit firmware.

As per direction (picture 3) I connected:

Shelly N to Gateopener CN3 -1
Shelly L to Gateopener CN3 -2
Shelly SW to a NC switch
Shelly I to Gateopener CN3 -6
Shelly O to Gateopener CN3 -5
-  NC switch (2nd cable) to Gateopener CN3-2 

Now the gate is opening correctly and closes correctly, however, the status is always shown as "closed". When the gate is moving, the status shows opening or closing, but the end state is always "CLOSED", regardless the gate is OPEN or CLOSED. When I move the magnet from the NC switch, the status remains the same (so actually the switch and magnet position doesn’t influence the status in Homekit and also not on its IP address.

[image]
[image]
[image]

I guess I did something wrong, but I can't figure what I did wrong. Any help is appreciated. TIA!

timoschilling, 2021-06-11:
You say that you use a 1PM, but your wiring is for a 1

TonyJacobi, 2021-06-11:
Title of picture 3 says: Shelly 1 or 1PM

timoschilling, 2021-06-11:
yes but the 1PM has a different pinout than the 1. The 1PM didn't have a I,  it was two L (L & L1) and the SW is on a different position.

TonyJacobi, 2021-06-12:
Thank you Timo, I guess switching the 1PM for a 1 will solve the problem.

timoschilling, 2021-06-12:
Post if you still have issues then

closed 2021-06-12 by timoschilling