mongoose-os-apps/shelly-homekit #704: Homekit shows Gate door Blocked

TonyJacobi, 2021-06-12:
I have installed the Shelly 1 to my gate opener. When I open the gate in Homekit, it shows opening and then "Opened". When I close the gate, the gate is closing and when it is closed, Homekit shows "Blocked" and in the Gate icon an orange exclamation mark.

I already changed the moving time, did not help. 

Any help on this issue is highly appreciated. TIA.

[image]

timoschilling, 2021-06-12:
Maybe your close sensors is misconfigured or didn't work well.

TonyJacobi, 2021-06-12:
Indeed Timo, changing it into normally open instead of normally closed, even though the switch is NC, solved the problem.

thanks!

closed 2021-06-12 by timoschilling