mongoose-os-apps/shelly-homekit #705: Garage Door Opener with Auto-Close Function

SMMMN, 2021-06-15:
Hi all, thanks for the great software. 

I live in a multi-family house and the garage can be opened with a remote control. The closing process happens automatically after 2 minutes.
I have two cables connected to the remote and I can open the garage in switch mode with the auto-off function. It would be practical if the Switch is now displayed as a garage so that I can also say "Open garage" to Siri. 

Is this already possible?

Many greetings
Simon

timoschilling, 2021-06-15:
We have support for garage door opener already. But no auto close functioning at the moment.

Pixel-Chris, 2021-06-24:
the most garade door opener have an integrated autoclose function
or you realise it in homekit with an automation

raddi, 2021-08-03:
+1 
A lot of car gates do have an auto close function, it would be nice to implement this feature to set a timer for closing without pressing a button :) 
Thanks!

andyblac, 2021-08-03:
can Auto Off be tweaked to be used for this ?

timoschilling, 2021-08-03:
Garage door has no auto off, but porting the auto off logic should be work

rojer, 2021-08-03:
yep, makes sense to add auto-close timeout.

raddi, 2021-11-24:
Anything new?

rojer, 2021-11-24:
nope