mongoose-os-apps/shelly-homekit #710: Shelly 2.5 Position Status say 100 but is 0 in real , also in Homekit the Roller is indicated vice versa

Marioishappy, 2021-06-17:
On my Shelly 2.5 the Position Status say 100 but is 0 in real , 
also in Homekit the Roller is indicated vice versa , it happened to me after i installed 3 new Switches , but only on 2 of the Shelly´s the fault is present. 
i tryed nearly everthing to change the indicated Status at Homekit, 
[image]

I tryed an Reboot, Reset, swap inputs swap outputs, also on the shelly all Connections are same as on the one which is working.

Marioishappy, 2021-06-17:
i managed to save another one by calibrating it, tryed this also before but again swapped the output cables and calibrated , now its good.

but one is left which start to calibrate and stop, the web ui is disconnected and after i refresh the page it is still uncalibrated.

Marioishappy, 2021-06-17:
now the last one is connected, but as soon as i press the physical switch or the web ui button the roller shutter is going only 1 cm down or up and stops directly again

rojer, 2021-06-17:
please open debug log in a separate tab (link in the "system" section), repeat your actions and attach the log here.

Marioishappy, 2021-06-17:
hey rojer, thanks
[image]

rojer, 2021-06-17:
that doesn't show all of it. use [gist](https://gist.github.com/) and copy/paste entire log capturing the device as it is doing the wrong thing.

Marioishappy, 2021-06-17:
21840775 shelly_debug.cpp:227    No log file, sending new entries
21898283 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 10.0.0.41:53575
22908062 HAPPlatformTCPStrea:121 0x3fff66a4 10.0.0.36:61284 Incoming HAP connection, ns 0/0/12
22918155 HAPPlatformTCPStrea:276 0x3fff66a4 10.0.0.36:61284 Accepted HAP connection, ns 0/1/12 ts 0x3fff662c
24059075 mgos_mongoose.c:66      New heap free LWM: 17464
24723793 shelly_main.cpp:543     Up 24.71, HAP 0/1/12 ns 1, RAM: 21324/38440; st 62; 4.1: c:1 mp:117.09 mt_ms:15070 cp:19.04 tp:19.04 lemd:0 lhmd:0
32715844 shelly_main.cpp:543     Up 32.70, HAP 0/1/12 ns 1, RAM: 22944/38440; st 63; 4.1: c:1 mp:117.09 mt_ms:15070 cp:19.04 tp:19.04 lemd:0 lhmd:0
40716350 shelly_main.cpp:543     Up 40.70, HAP 0/1/12 ns 1, RAM: 22944/38440; st 63; 4.1: c:1 mp:117.09 mt_ms:15070 cp:19.04 tp:19.04 lemd:0 lhmd:0
48716430 shelly_main.cpp:543     Up 48.70, HAP 0/1/12 ns 1, RAM: 22944/38440; st 63; 4.1: c:1 mp:117.09 mt_ms:15070 cp:19.04 tp:19.04 lemd:0 lhmd:0
56091211 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 10.0.0.41:53575
56716045 shelly_main.cpp:543     Up 56.70, HAP 0/1/12 ns 1, RAM: 22944/38440; st 63; 4.1: c:1 mp:117.09 mt_ms:15070 cp:19.04 tp:19.04 lemd:0 lhmd:0
57072680 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 10.0.0.41:53575
58076000 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 10.0.0.41:53575
59079389 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 10.0.0.41:53575
60084314 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 10.0.0.41:53575
64715979 shelly_main.cpp:543     Up 64.70, HAP 0/1/12 ns 1, RAM: 22944/38440; st 63; 4.1: c:1 mp:117.09 mt_ms:15070 cp:19.04 tp:19.04 lemd:0 lhmd:0
72715918 shelly_main.cpp:543     Up 72.70, HAP 0/1/12 ns 1, RAM: 22944/38440; st 63; 4.1: c:1 mp:117.09 mt_ms:15070 cp:19.04 tp:19.04 lemd:0 lhmd:0
80715500 shelly_main.cpp:543     Up 80.70, HAP 0/1/12 ns 1, RAM: 22944/38440; st 63; 4.1: c:1 mp:117.09 mt_ms:15070 cp:19.04 tp:19.04 lemd:0 lhmd:0
85072994 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 10.0.0.41:53575
86109315 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 10.0.0.41:53575
87081105 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 10.0.0.41:53575
88080501 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 10.0.0.41:53575
88716264 shelly_main.cpp:543     Up 88.70, HAP 0/1/12 ns 1, RAM: 22944/38440; st 63; 4.1: c:1 mp:117.09 mt_ms:15070 cp:19.04 tp:19.04 lemd:0 lhmd:0
89099140 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 10.0.0.41:53575
89135004 mgos_mongoose.c:66      New heap free LWM: 17304
90082562 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 10.0.0.41:53575
91085713 mg_rpc.c:305            Shelly.GetInfoExt via WS_in 10.0.0.41:53575
96716266 shelly_main.cpp:543     Up 96.70, HAP 0/1/12 ns 1, RAM: 22944/38440; st 63; 4.1: c:1 mp:117.09 mt_ms:15070 cp:19.04 tp:19.04 lemd:0 lhmd:0
104716755 shelly_main.cpp:543     Up 104.70, HAP 0/1/12 ns 1, RAM: 22944/38440; st 63; 4.1: c:1 mp:117.09 mt_ms:15070 cp:19.04 tp:19.04 lemd:0 lhmd:0
112716626 shelly_main.cpp:543     Up 112.70, HAP 0/1/12 ns 1, RAM: 22944/38440; st 63; 4.1: c:1 mp:117.09 mt_ms:15070 cp:19.04 tp:19.04 lemd:0 lhmd:0
115416745 shelly_input.cpp:76     Input 1: change (state 1)
115424345 shelly_hap_window_c:342 WC 1: Tgt pos 19.04 -> 100.00 (ext)
115430771 shelly_hap_window_c:322 WC 1: State: idle -> move (0 -> 20)

Marioishappy, 2021-06-17:
so for me it seems like as soon as i press one button the shelly is rebooting

Marioishappy, 2021-06-17:
i am also not able to install 2.9.0 or reinstall 2.9.1

Marioishappy, 2021-06-17:
as whenever I do any action the shelly is rebooting the only thing I was able to do is an 10 Sek reset at the shelly it self, thereafter http://192.168.33.1 web ui had the same issues as before.. how can i revert to stock or 2.9.0 or remove all settings to start again. all possibilitys to revert are lost due to the rebooting

timoschilling, 2021-06-17:
Looks like you are trying to revert in the access point mode that’s not possible
1. the browser isn’t able to load the stock firmware when connected to the Shelly AP
2. the Shelly didn’t accept updates in the AP mode

Marioishappy, 2021-06-20:
Managed after several resets and revert attempts to install stock fw , for now it looks like the reboot issue happens at stock fw as well. 
I just ordered a new Shelly and will see if the issue persists.

New shelly arrived and installed, stock fw works fine ... also the Position indication.

Installed the HomeKit fw and calibrated. Had to switch the output button at the web ui.
Again the Position which is indicated at the home app is wrong. Even after calibration.

Marioishappy, 2021-06-20:
At the stock fw there is a switch to change the status ( position it is shown in the app ) wasn't that the case either for the HomeKit fw before the 2.9.1 ?

Marioishappy, 2021-06-21:
c/out an reset ( 10 sek at the shelly back ) and pressed the calibration button several times afterwards. Now the indication is correct.
Thanks for the Support and Best Regards.

closed 2021-06-21 by timoschilling