mongoose-os-apps/shelly-homekit #712: advise on how to use shelly-homekit with current Hue bulbs using the shelly switch

zeltak, 2021-06-21:
Hi guys

so i started with home automation a year ago and at the time bought a lot of phillips Hue bulbs. for several reasons i decided to move over all the house to use shelly-homkit and very please so far

the only issue i cant figure out yet is how to make my current bulbs work with shelly. for example in the cupboard i have a shelly2.5 that is connected to a switch that turns lights on/off in the kitchen and cupboard. in the cupboard the switch works great to turn off/on the hue light but it seems like when the switch is in kept in the off state (my wife... ;-) ) then the hue bulb won't work and turn on (via the homekit\hue app). this is especially annoying as i have a hue motion sensor that auto activates the light in the cupboard which again dosent work when the switch is of..

i tried the `toggle` and  `edge` switch options but non seem to work. any advice on how to approach this?

thanks so much in advance

Z

timoschilling, 2021-06-21:
Set the input type to detached, reload / reboot the UI, then you have a stateless switch in HomeKit which you can configure to turn on the hue if switch is used. Keep in mind that the hue must be continued be powered

zeltak, 2021-06-22:
thx @timoschilling , appreciate the response!

that seems to kill the physical switch altogether (unless im missing something?) and would cause my wife to throw me out of the house :) (..she is already upset with all my home automation experiments..lol).
basically i want the physical switch that has a on/off mode (button up/down) to work as usual (using edge i guess) but also when the lights are off to be able to still have the hue activate via the motion sensor. 
does that make any sense :D 

thx

Z

timoschilling, 2021-06-22:
please read https://github.com/mongoose-os-apps/shelly-homekit/wiki/Stateless-Toggle-Switch

mvnixon, 2021-07-08:
Hi there,

Trying to do this myself - I have a switch outside connected to the Shelly 1L, which controls the outdoor lights the Shelly unit is plugged into. But I also want the switch to control my other outdoor Hue lights via HomeKit. Stateless doesn't seem to be the solution here as the physical switch becomes defunct. It would be good to have the switch appear in HomeKit as a switch, which could then also be set up to turn on/off the other lights. 

I read the guide above - but this doesn't seem to help in this circumstance. Any ideas?

Dabr422, 2021-07-10:
When configuring the stateless switch in HomeKit you can trigger any combinations of devices or scenes.

mvnixon, 2021-07-10:
thank you - understand that, but is there a way to have the physical switch, connected to sw1 on the shelly, show up in HomeKit? I.e. physical wall switch is toggled, triggers a combination of devices/scenes.

Dabr422, 2021-07-10:
The physical button is exactly what shows up in HomeKit in detached mode. But if it’s the toggle function you’re looking for the answer is no as far as I know. I use it with a spring loaded button which makes it easy to do single, double or long press.

mvnixon, 2021-07-11:
Thanks, yes if using it in detached mode, it renders the physical toggle switch defunct unfortunately.

jonasman, 2021-07-13:
@mvnixon it does not. The physical switch can be configured to run automations in home kit. I just configured mine like that.
You set it up as a momentary button, then on home app, add a configuration for the single press. I added a Shortcut to toggle the lamp like the docs say https://github.com/mongoose-os-apps/shelly-homekit/wiki/Stateless-Toggle-Switch

closed 2021-08-21 by github-actions[bot]