mongoose-os-apps/shelly-homekit #713: Support magnetic sensor as add-on on Shelly 1 for Garage door mode

vistalba, 2021-06-22:
I use Shelly 1 with garage door mode for my 24VDC garage door.
Shelly is powered by the 24VDC from the motor unit directly.
I added on magnetic contact to detect when door is closed.
Now I want to add another magnetic contact to also get open state with Shelly 1 Add-on.
As I can see this isn't supported in the UI. (Or at least it doesn't show up on my test shelly).

Any Plans to make this possible?

So the SW is the "closed" state sensor and the Add-on sensor will report the open state like Shelly 2.5 does?

timoschilling, 2021-06-22:
I think thats an interesting feature.

vistalba, 2021-06-22:
Yes. I can't use Shelly 2.5 as it isn't "potentialfrei" so the options I have at the moment is:

- Use Shelly 2.5 with a additional relais
- Use Shelly 1 with just one contact sensor 

I do not like both of them. Then I saw, that this Add-on with magnetic contact is already implementet in Shelly stock firmware. So here we are :D

Pixel-Chris, 2021-06-24:
+1   Add-on support !!!  for shelly 1

Pixel-Chris, 2021-09-07:
Is there some time for that left?

Pixel-Chris, 2021-09-07:
See #186 also

daitarn31974, 2021-12-13:
+1 here too.
I would like to have the opportunity to use addon on a shelly1 so that I would be able to use them as contact sensors for home doors running Shelly with 220V and not running power on the magnetic sensor. Actually it is possible only in 12/24V but I do not want to deploy 5/6 transformers around home

Pixel-Chris, 2022-03-17:
Is that working now?

timoschilling, 2022-03-17:
It will work when this issue is closed.

Pixel-Chris, 2022-03-17:
Temperature add-on is not the same?

timoschilling, 2022-03-18:
@Pixel-Chris it's the same hardware add on, but on the software side that are two different things.

marcus-aa, 2022-06-18:
This would be a great door opener feature. Two sensors, one dry relay.  Exactly what most door openers need.

Pixel-Chris, 2022-06-21:
Maybe the next…? 😉

markirb, 2022-11-24:
this is now in a PR, where you can find a Testing version for Shelly1, if someone wants to try it out https://github.com/mongoose-os-apps/shelly-homekit/pull/967

vistalba, 2022-11-25:
@markirb How do I get the ZIP file to flash the Shelly?

markirb, 2022-11-25:
upload on webinterface

vistalba, 2022-11-25:
Lol, you get me wrong. Sorry for that. Where can I dpwnload the ZIP to flash it?

markirb, 2022-11-25:
Haha I wondered 😂 https://github.com/mongoose-os-apps/shelly-homekit/pull/967#issuecomment-1321210345

closed 2024-07-22 by markirb