mongoose-os-apps/shelly-homekit #714: Mistake happend....  Password protection set and revert to an older FW

Pixel-Chris, 2021-06-24:
Hello, i update my garage door shelly 2.5  to the latest 

i set an password ( i still know it  ;-) )
tested it

and then i revert to 2.73 firmware
no i can't access the webgui
it ask for an user an the password

what user must be used?

Thanks

Pixel-Chris

rojer, 2021-06-24:
hm, that's bad news i'm afraid. going into failsafe mode performs a full configuration reset, that should help.
problem is that before 2.9.1 it's difficult to trigger on 2.5 because of the internal button arrangement.
you'll need to press the external button immediately after applying power. might take a few tries...

timoschilling, 2022-04-29:
closed because of inactivity

closed 2022-04-29 by timoschilling