mongoose-os-apps/shelly-homekit #715: Chime + Digital switches in doorbell mode

rmendes22, 2021-07-01:
> But I guess since there is no UI that means there is no switch for digital and mechanical gong like in the screenshot @DerKoller-eu posted?

The switch is created by the Homebridg plug-in. S as this is a nice feature I would love to see this in a Shelly HomeKit Firmware as well..

_Originally posted by @DerKoller-eu in https://github.com/mongoose-os-apps/shelly-homekit/issues/375#issuecomment-778570114_

Hi all,

First of all, many thanks for the great work made in this project!  💪

Would it be possible to implement the suggestion made by @DerKoller-eu in #375? i.e., implement something like this Homebridge plugin? https://gitlab.com/ca-iot/homebridge-shelly-doorbell

I managed to get a doorbell working (notification on iPhone + chime on HomePod) but I cannot create an automation to also trigger the Shelly switch and ring the analog doorbell and also I cannot schedule to disable it after a certain hour.

I modified the configuration to work as a detached switch and I then managed to get a switch and a button. On the button press I receive two notifications (button closed + button opened) and I also manage to create an automation that triggers the Shelly switch to ring the analog doorbell for 0.6s (using the auto-off in the Shelly config). This way I can also disable the analog switch at night. But I don't get the chime on HomePod and the duplicated notifications for each doorbell button press are a small nuisance.

It would be great to have the best of both worlds: a "real" doorbell with a notification and HomePod chime and two switches to disable the analog chime and the digital one.

Is this possible?

Thanks in advance!

appliko, 2021-07-02:
I add myself to this request.
Actually I cannot add an I3 into my configuration in doorbell mode.
I added as a motion sensor, and I use it to trigger doorbell dummy switch from ffmpeg camera in homebridge.
It works, but I would prefer to don't depend by homebridge, if possible

timoschilling, 2021-07-02:
@appliko can you add a picture of your shelly settings and of the HomeKit error messages please

appliko, 2021-07-05:
> @appliko can you add a picture of your shelly settings and of the HomeKit error messages please

Update: it was my fault. I did try to add the switch instead of the entire I3, picking it up from previous devices list.
I restart the server on I3 and I reconnect it, and it works. It show the switch as non compatible in the main page but I got the doorbell events, connected to the ffmpeg camera in same room
Sorry it takes a while to answer but when I change the configuration my homekit crash and I had to setup everything again

timoschilling, 2021-07-05:
That the doorbell is show as not supported is expected behavior, don’t worry about that it work anyway.

appliko, 2021-07-05:
> That the doorbell is show as not supported is expected behavior, don’t worry about that it work anyway. It’s a strange behavior of the home app, maybe because it’s deprecated. But is definitely not a bug on our side, the same happens with a doorbell from the apple HomeKit simulator

I don't worry, it is expected, as I read on other comments. It works well and I'm happy !

rmendes22, 2021-07-05:
> That the doorbell is show as not supported is expected behavior, don’t worry about that it work anyway. It’s a strange behavior of the home app, maybe because it’s deprecated. But is definitely not a bug on our side, the same happens with a doorbell from the apple HomeKit simulator

My "issue" is not that it shows as unsupported, it's that there is no "switch event" (don't know if it's the right term) attached to it that can be used to trigger other events, for example, when the doorbell rings turn on a light along with the automatic chime on HomePod and the notification on iPhone.

From what I could understand from the homebridge-shelly-doorbell plugin not only that should be possible but the doorbell chime itself can be disabled, something that isn't possible with the current HomeKit Shelly firmware.

appliko, 2021-07-05:
> My "issue" is not that it shows as unsupported, it's that there is no "switch event" (don't know if it's the right term) attached to it that can be used to trigger other events, for example, when [...]

Just as a patch waiting for new features, using an I3, why don't you use two inputs in parallel ? Then you could keep one input as doorbell and the other as a motion sensor with events you can use for automation

markirb, 2022-10-01:
Now that I also made my doorbell smart with a Shelly UNI, I would also love to see this feature.

I was thinking about the implementation a little bit:
When having a Doorbell (Button and Gong) and the Input is set to Detatched and HAP-Type Doorbell, we could then create two virtual Switches additional to the Switch (Physical Gong), where one of those controls passing events to the Homekit Gong and the other one to the physical gong (Switch).

I also have the problem that my phyisical doorbell is bouncing quite a lot, which leads to double events most of the time.
I think this would be gone when Doorbell derives from "SensorBase" and not from "StatelessSwitchBase". As far as I saw you cannot handle different events for the doorbell anyway.

Any thoughts on this?

5xPa, 2022-10-01:
I had the same problem of "bounce" I solved this with a cheap timer. When the door bell is pushed the timer relay changes stated and holds that state for 60 seconds.

markirb, 2022-10-01:
Sure I could get away with a cheap Timer, but changing it to a Sensor would do exactly that, with the benefit of not adding any code, be adjustible for everyone via web app, if I did not miss anything here

markirb, 2022-10-02:
Ok after looking at it cannot be a Sensor. But I think adding some debouncing Logic would make sense in general. Nobody can hit a physical switch as fast as 20ms i guess.

closed 2022-11-10 by github-actions[bot]