mongoose-os-apps/shelly-homekit #716: Shelly device with multiple accessories and serial numbers

andyblac, 2021-07-02:
It seems that each accessory within a single device should be using a unique serial number.

[image]

Can we add add '-?' ? being the accessory ID, to the main serial number for each accessory?

This should help with compatibility with other HomeKit apps

rojer, 2021-07-03:
i guess we can amke it even more fancy, e.g. `${model}-${mac}-${aid}`
it's simple enough, would you like to give it a go yourself?

andyblac, 2021-07-03:
Atm I have no computer I had to return my new iMac as lan port was faulty, won't get replacement till 2 August

rojer, 2021-07-03:
ah, i see. no problem, made the change myself.
serial number will now include -${aid} at the end. i decided no to include model, after all.

closed 2021-07-03 by rojer

rojer, 2021-07-03:
pushed 2.9.2-beta1